lgsvl/simulator #1051: What preparation beforehand is needed for editing HD map of Shalun?

kotakky, 2020-10-09:
I'm studying the HD map's operation by tutorial.
I cannot understand the procedure of HD map's edition and so, I contact you.
I want to know what preparation beforehand is needed for editing HD map of Shalun that is defalt data ,and know about the procedure.
If possible, please tell me the video, youtube  etc… .

luqiang21, 2020-10-09:
@kotakky You can follow tutorials here: https://www.lgsvlsimulator.com/docs/map-annotation/ to learn how to annotate each element in the map. Let me know if you have any other questions when reading the document.

kotakky, 2020-10-12:
Thank you,luqiang21.
I cannot understand the procedures that are marked by red cycle in attached photo.

In order to annotate HD map ,is it needed that connecting autoware with lgsvl by clonning https://github.com/lgsvl/Autoware?

I downloaded lgsvlsimulator-windows-64-2020.06 , from https://github.com/lgsvl/simulator/releases/tag/2020.06.
[image]

EricBoiseLGSVL, 2020-10-12:
@kotakky 
- Default layer is set in the Inspector panel in the Unity Editor.  All imported meshes default to this layer but in case a user changes this or imports a purchased asset that has changed this, we want the layer set to Default.
- Road meshes need a MeshCollider Unity component for physics to work in the simulation.  You can use any collider but we recommend mesh because it sets to the shape of the mesh.  Please read Unity docs on mesh collider [here](https://docs.unity3d.com/Manual/class-MeshCollider.html).

closed 2020-10-21 by kotakky