lgsvl/simulator #1052: How to control reverse motion of an ego vehicle?

User-712, 2020-10-11:
Hi LGSVL Team,

I can control the forward motion and steering of an ego vehicle using twist_cmd as shown below 
rostopic pub -r 10 /vehicle_cmd autoware_msgs/VehicleCmd '{twist_cmd:{twist:{linear:{x: 5.0},angular: {z: 0.2}}}}

but how about the other functionalities? 
Could you please clarify how the vehicle could be controlled in reverse motion?

Thanks!

EricBoiseLGSVL, 2020-10-12:
@User-712 We do not have reverse control with ROS, you will need to add this logic in the VehicleControlSensor.

User-712, 2020-10-12:
@EricBoiseLGSVL  could you please explain in brief what all has to be done in order to control gear or reverse motion of an ego vehicle through ROS using  /vehicle_cmd autoware_msgs/VehicleCmd or if there is any other msg format or topic to control the same.

It would be very helpful if I get the procedure. 

Thanks!

EricBoiseLGSVL, 2020-10-12:
Please look at the VehicleControlSensor.cs.  Here you can see the AD stack type and how the sensor applies changes to the VehicleDynamics component depending on the commands sent.  You may need to add the topics to the sensor also.  Then get the dynamics component and set the vehicle to reverse = true.  Now the vehicle will be in reverse gear.  @hadiTab anything I am missing?

EricBoiseLGSVL, 2020-10-20:
Please see issue #1062 for help with this

closed 2020-10-20 by EricBoiseLGSVL