lgsvl/simulator #1054: Lidar Pattern Data

osholsonel20, 2020-10-12:
Currently you have the velodyne setup for Lidar, however, do you have any direction or ideas of if I had specific points that I would like to loop through while still using the GPU async setup? I have timestamps and spherical coordinates in an excel document. I haven't given the lidar script a look in sometime, but if you had any pinpoint solutions that I should take a peak at, that'd be great!

EricBoiseLGSVL, 2020-10-12:
@osholsonel20 Can you clarify what you are trying to do and what the issue is please?

osholsonel20, 2020-10-12:
Iterating through a list of custom vectors for lidar raycasts vs. how you have it setup today.

rongguodong, 2020-10-12:
> Iterating through a list of custom vectors for lidar raycasts vs. how you have it setup today.
> 
What does it mean?

If you want to know the point pattern of the Velodyne Lidar Sensor plugin, please refer to Velodyne's user manual. We precisely follow the pattern of the real Lidar.

osholsonel20, 2020-10-12:
I have mapped out the scan pattern from a LiVOX Horizon Lidar and have the point pattern, do you have a way of plugging this into your lidar solution? Velodyne is a great lidar sensor, but there are also others. I really want the performance benefits that you have with your GPU/Shader lidar solution, but with my scan pattern. Trying to find where to hook on to the lidar script, or will I have to re-write most of it?

rongguodong, 2020-10-12:
To support other types of Lidar sensors, you need to build your own Lidar sensor plugin. Please refer to our document about that:
https://www.lgsvlsimulator.com/docs/lidar-plugin/#build-your-own-lidar-sensor-plugin

At this time, the Velodyne Lidar Sensor Plugin is not open sourced.

EricBoiseLGSVL, 2020-10-12:
@osholsonel20 yes, @rongguodong is correct.  The Velodyne Lidar was created to support the sensor specs and is not open source but our current Lidar sensor is similar.  You will need to look at the specs for LiVOX Horizon Lidar sensor and create a new sensor class.  I'm sure the community would appreciate new real world sensors. :)