lgsvl/simulator #1056: Unable to connect Autoware.auto and LGSVL

MaxandreOgeret, 2020-10-12:
Hello everyone, I am trying to run LGSVL alongside autoware.auto but I am totally unable to make anything work. I am following thoses tutorials :
 - https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html
 - https://www.lgsvlsimulator.com/docs/autoware-auto-instructions/

### Using `ROS2 native`

In the first tutorial it is said to download the latest assets and setup the bridge as ROS2 native. But when starting the simulation with ROS2 native the simulator crashes with this error message in the command line : `Aborted (core dumped)`
And this error message in the web UI : `Invalid: type_support is null, at /tmp/binarydeb/ros-dashing-rmw-cyclonedds-cpp-0.5.1/src/rmw_node.cpp:1311, at /tmp/binarydeb/ros-dashing-rcl-0.7.9/src/rcl/publisher.c:171`

### Using `ROS 2 bridge`.

I set the bridge to `ROS2` in the vehicle, I start the simulation and the it starts well. Then I start the bridge with `lgsvl_bridge` but the bridge reports this error : `[ERROR] [lgsvl-bridge]: Unknown operation received from client (71), disconnecting client` And it's not working. 

Any idea here ? Thanks

hadiTab, 2020-10-12:
@MaxandreOgeret We do not support the ROS2 native implementation as it was not developed by us and we do not have access to the source code. I would suggest asking the autoware team about this. According to their [troubleshooting page](https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/troubleshooting.html) it seems that they suggest posting questions to ROS answers and tagging the questions with Autoware. The Autoware team watch the tag and will respond.

The ros2-lgsvl-bridge only works with release 2020.06 or later. The binary provided in the autoware-auto containers is based on 2020.05 and does not support the ros2-lgsvl-bridge. If you want to the binary provided with autoware-auto with a bridge you can use it with the ros2-web-bridge (but it will have performance issues with a standard sensor arrangement).

gurbain, 2020-12-01:
Hi,

Maybe I'm wrong but it looks like the error message:

> [ERROR] [lgsvl-bridge]: Unknown operation received from client (71), disconnecting client

could be due to a ROS1-ROS2 bridge uncompatibility. By default, lgsvl-bridge uses ROS2-bridge protocol, whereas the default autoware vehicle is configured for ROS1. Please check issue #1015 or [vehicle configuration documentation](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle) if this can help you.

MaxandreOgeret, 2020-12-01:
Hey ,  no the car was configured correctly. I was using the Autoware fork of LGSVL and I had this error because autoware auto was  not wourced.

gurbain, 2020-12-01:
Ok my mistake then! So maybe the issue should be closed?

endland, 2020-12-07:
Did you clear this problem? 
In my case, Rviz seem to word properly, but "lgsvl_bridge" has a problem. 

_

> $ rviz2 -d /home/"${USER}"/AutowareAuto/install/autoware_auto_examples/share/autoware_auto_examples/rviz2/autoware.rviz
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-kzi'
> 1607337040.621109 [0]      rviz2: using network interface wlp69s0 (udp/192.168.0.181) selected arbitrarily from: wlp69s0, docker0
> [INFO] [rviz2]: Stereo is NOT SUPPORTED
> [INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
> [INFO] [rviz2]: Stereo is NOT SUPPORTED

_

> $ lgsvl_bridge 
> 1607337529.307688 [0] lgsvl_brid: using network interface wlp69s0 (udp/192.168.0.181) selected arbitrarily from: wlp69s0, docker0
> terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
>   what():  bind: Address already in use
> Aborted (core dumped)

Any advice would be helpful? Thanks.

MaxandreOgeret, 2020-12-07:
I dont think those errors are related with my problem, but 
- Do you start ade with the file `.aderc-lgsvl` ?
- This means that the port is already in use

endland, 2020-12-08:
Thanks. Your comment gave me a clue. I checked if the port is used. Oddly, It is already used by gitlab. I changed the port number in webUI of simulator. It works.

guilleherreraf, 2021-01-22:
Hello guys! I'm attempting to run the Autoware.Auto stack installed from source on a Jetson Xavier NX (Ubuntu 18.04) alongside the LGSVL simulator installed in a different machine running Windows 10. I have both platforms connected through a Gibabit ethernet switch. 
I'm comparing the process to do so between the [LGSVL simulator](https://www.lgsvlsimulator.com/docs/autoware-auto-instructions/) documentation and the [Autoware demos](https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html) and they have different instructions to follow. 

- The LGSVL simulator documentation states that we must install `ros2_lgsvl_bridge` outside the Autoware.Auto workspace, and then install the `lgsvl_msgs` package inside the AutowareAuto workspace. 
- On the other hand, in the Autoware demos they use the `lgsvl_interface` package. 

Which of both implementations should I follow, considering my set-up? Also, I'm guessing that the `ros2_lgsvl_bridge` should be installed in the Xavier NX and by simply configuring the port and the IP in the LGSVL's GUI this should work?

ZhenshengLee, 2021-03-16:
@guilleherreraf the default `lgsvl_interface` doesn't work but `ros2_lgsvl_bridge` works.
In my environment(ubuntu1804+ros2-dashing+cuda10.2+lgsvl202006), ros2 doesn't get any msg from `lgsvl_interface`.
So maybe we have to adjust the code in `lgsvl_interface` to get the avp demo running properly.