lgsvl/simulator #1057: LGSVL does not start on Ubuntu 18.04.

MaxandreOgeret, 2020-10-12:
When starting the pre-compiled version downloaded from the release section it fails to start. I only get those messages and nothing else : 

```
$ ./simulator          
Set current directory to /home/maxandre/Downloads/lgsvlsimulator-linux64-2020.06
Found path: /home/maxandre/Downloads/lgsvlsimulator-linux64-2020.06/simulator
```

A colleague at work has the same problem. Any idea ? 
Thanks

EricBoiseLGSVL, 2020-10-12:
@MaxandreOgeret please post the player.log to help debug the issue.

duongnxptit, 2020-10-23:
> @MaxandreOgeret please post the player.log to help debug the issue.

i have same problem like @MaxandreOgeret . Can you tell me where i can find  log file to show it for you?

EricBoiseLGSVL, 2020-10-23:
@duongnxptit and @MaxandreOgeret Player.log [here](https://docs.unity3d.com/Manual/LogFiles.html)

MaxandreOgeret, 2020-10-23:
@EricBoiseLGSVL Here's the content of the file :+1: 

```
Mono path[0] = '/home/maxandre/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
Mono config path = '/home/maxandre/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Loading player data from /home/maxandre/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 60 Hz
Vulkan detection: 0
No supported renderers found, exiting
(Filename:  Line: 618)
```

Nvdia drivers are installed properly and working. Docker version works fine also.

EricBoiseLGSVL, 2020-10-23:
@MaxandreOgeret Thanks for the log, this helps.  Looks like you need to install the nvidia drivers and force simulator application to use the GPU.  Please look at similar issue posts about like #1076 #879 #753 and documentation [here](https://www.lgsvlsimulator.com/docs/getting-started/)

closed 2020-12-02 by EricBoiseLGSVL