lgsvl/simulator #1058: Training is suspended due to network interruption during training

xiao-hua-sheng, 2020-10-13:
After training for a long time, the simulator screen is still and pycharm has no printing error. At this time, the network is disconnected, and the training can not be recovered normally after reconnecting to the network. Does the simulator need to log in regularly to confirm the account information? How to deal with this problem?

EricBoiseLGSVL, 2020-10-13:
@xiao-hua-sheng This should not happen.  We will look into this.  What was the length of time before disconnection?

xiao-hua-sheng, 2020-10-14:
After more than 50 hours of training

EricBoiseLGSVL, 2020-10-20:
Do you have any errors in the log?  We are having issues reproducing this.

xiao-hua-sheng, 2020-10-20:
There is no error output, it may also be that my system runs for too long. I'm still training normally.