lgsvl/simulator #1059: Method to get Apollo

johnjerome1992, 2020-10-13:
I have the below setup: 
LGSVL simulator running with Apollo 5.0 communicating with CyberRt Bridge. 
I am using Python API to create test scenes and see the behavior of the apollo autonomous driving stack. 

I would like to know if there is a way to get the data published from Apollo to Simulator via the topic "/apollo/control"
Basically, I would like to store or record the control data sent from apollo to LGSVL using the python API. 

If this is not possible then can you please help me with an alternative approach?
Thanks

lemketron, 2020-10-13:
Have you tried the [cyber_recorder utility](https://cyber-rt.readthedocs.io/en/latest/CyberRT_Developer_Tools.html#cyber-recorder)? As you noted, Apollo sends the control commands to the simulator over CyberRT and they are not available in the Python API. If you're just trying to store/record the control data, the easiest way to do that would be with cyber_recorder.

johnjerome1992, 2020-10-13:
Thanks for the response.
Yes, I have tried it. But it has to be run from the apollo docker environment and can not be easily read from other tools. 
I wanted to have control over this recording. I mean, I wanted to record only when I run a test scene and also not the continuous data only at certain periodic intervals. 
Also, I am not sure how to initiate this cyber_recorder that is inside the apollo docker from outside or from my python script running on the host PC.

Do you think is there a way to create a passive cyber RT node outside the apollo docker environment and control this from a custom script to just subscribe to the topic "/apollo/control" same as how LGSVL simulator is doing in ego vehicle control.

ntutangyun, 2020-10-14:
Hi @johnjerome1992 maybe you could try out the cyber RT python API. i.e. Create a python script with the API, subscirbe to the channel you want, and listens to the outside commands as well.

rongguodong, 2020-10-14:
Apollo has [Cyber RT Python API](https://github.com/ApolloAuto/apollo/tree/master/cyber/python). You can use Cyber API to read any Cyber channel to your Python script.

johnjerome1992, 2020-10-16:
@ntutangyun and @rongguodong thanks for the information. I will try them.

Yimags, 2020-12-10:
> @ntutangyun and @rongguodong thanks for the information. I will try them.

Hello mate, I am also trying to get and store those data from cyberRT. Could you give some hints about how to get them with pythonAPI? Thx in advance!

closed 2021-08-05 by EricBoiseLGSVL