lgsvl/simulator #1061: Issue in installing Unity 2019.3.3f1 on Ubuntu 16.04

User-712, 2020-10-14:
Hi Team, 

I tried installing Unity Editor in Unity Hub, but couldn't find Unity 2019.3.3f1 when I execute <PATH_TO_UNITY_HUB> unityhub://2019.3.3f1/7ceaae5f7503 .
[image]
Can I install latest version of Unity?
Or  how can I install Unity 2019.3.3f1?

I followed below procedure:
[https://www.lgsvlsimulator.com/docs/build-instructions/](url)

rongguodong, 2020-10-14:
Please click "download archive" to find 2019.3.15f1. (The instruction is a little bit out dated. For 2020.06 simulator, you need Unity 2019.3.15, not 2019.3.3)

EricBoiseLGSVL, 2020-10-14:
@User-712  Yes use the download archive for 2019.3.15f1 for branch 06 release, not master but the branch.

User-712, 2020-10-15:
thanks for your quick reply, issue is resolved : )

closed 2020-10-15 by User-712