lgsvl/simulator #1062: Adding reverse control to the Simulator with ROS.

Mojojojooo, 2020-10-15:
Hey, @hadiTab @EricBoiseLGSVL I still have some doubts please help me out:

1.Code:
controlType = ControlType.AutowareAi;
if (data.ShiftGearUp || data.ShiftGearDown)
{
if (data.ShiftGearUp) Dynamics.GearboxShiftUp();
if (data.ShiftGearDown) Dynamics.GearboxShiftDown();

ADAccelInput = data.Acceleration.GetValueOrDefault() - data.Breaking.GetValueOrDefault(); // converted from lin accel 
ADSteerInput = data.SteerAngle.GetValueOrDefault(); // angle should be in degrees
 }
 else
 {
 if (Dynamics.Reverse) return; // TODO move?

Is this the part of the code which needs to be modified so that it subscribes to the data and the Reverse status is updated?

2. If the above part is where the changes need to be made can anyone help me with adding it ? as I have no experience in C# as well as Unity.

Thanks in advance.

EricBoiseLGSVL, 2020-10-15:
@Mojojojooo Yes if you are using AutowareAi then this is the place.
- Get the topic data, e.g. if (data.Reverse) (not sure if this is in the conversion data class, you may need to edit there also)
- Then if true Dynamics.Reverse = true;
- You probably also want to be able to go back to first gear so the if (data.ShiftGearUp) will need to check if (Dynamics.Reverse) and set Dynamics.Reverse = false; and Dynamics.GearboxShiftUp; else just Dynamics.GearboxShiftUp;

Make sense?

Mojojojooo, 2020-10-16:
Hey, @EricBoiseLGSVL I tried making changes as you said :
1. In VehicleControlSensor.cs
Made changes to try
                        if (data.Reverse) Dynamics.Reverse = true;
                        if (data.ShiftGearUp) {
                            if(Dynamics.Reverse) {
                                Dynamics.Reverse = false;
                                Dynamics.GearboxShiftUp();
                            }
                            else {
                                Dynamics.GearboxShiftUp();
                            }

                        }
2. In VehicleControlData:
         // autoware
        public float? Velocity;
        public float? SteerAngularVelocity;
        public float? SteerAngle;
        public bool ShiftGearUp;
        public bool ShiftGearDown;
        public bool Reverse; //Added a reverse variable 
Now I unity I am getting the following error:
[image]

Please help me with this.

EricBoiseLGSVL, 2020-10-16:
@Mojojojooo 
- First off, did you change RosConversions.cs (ln 679) and Autoware.cs (ln 53) code for this new variable?  You will need to make sure simulator can parse the data from the topic vehiclecmd to set Reverse.
- Second, just look at IVehicleDynamics.cs.  It has ShiftReverse() method, call that not Dynamics.Reverse = false;

Mojojojooo, 2020-10-19:
Thanks for the help @EricBoiseLGSVL.  I am able to put it into reverse now. But the car seems to be not braking when the acc is removed. It just starts going in reverse while increasing its speed rapidly. Can someone help me with this issue?

EricBoiseLGSVL, 2020-10-19:
@Mojojojooo No problem, glad to help.  Please check that the control sensor is reverting the acceleration to 0.  Is it sending 0 or are you just not sending input anymore?  If you are not sending you will need to be sure to set accel and brake to 0 before evaluating the command value, that way it sets to 0 unless getting a value.  Make sense?

Mojojojooo, 2020-10-20:
Hi, @EricBoiseLGSVL below is the part of code that was there before.
                        **CODE :** var linMag = Mathf.Clamp(Mathf.Abs(data.Velocity.GetValueOrDefault() - ActualLinVel), 0f, 1f);
                        ADAccelInput = ActualLinVel < data.Velocity.GetValueOrDefault() ? linMag : -linMag;
                        ADSteerInput = -Mathf.Clamp(data.SteerAngularVelocity.GetValueOrDefault() * 0.5f, -1f, 1f);
I am not understanding what it does and wanted to know if this is causing the issue of rapid increase in speed?

Should I replace it with this part which is available in the forward gear part .
**CODE:**                         ADAccelInput = data.Acceleration.GetValueOrDefault() - data.Breaking.GetValueOrDefault(); // converted from lin accel 
                        ADSteerInput = data.SteerAngle.GetValueOrDefault(); // angle should be in degrees

EricBoiseLGSVL, 2020-10-20:
- I was wrong about the setting to 0, this is already done in Update() here.  After getting no more commands it defaults to 0.
```
// LastControlUpdate and Time.Time come from Unity.
if (SimulatorManager.Instance.CurrentTime - LastControlUpdate >= 0.5)    // > 500ms
{
    ADAccelInput = ADSteerInput = AccelInput = SteerInput = 0f;
}
```
- Did you remove the if reverse return?
- Please debug ADAccelInput, linMag and ActualLinVel.  It is difficult to debug without posting values
```
ADAccelInput = ActualLinVel < data.Velocity.GetValueOrDefault() ? linMag : -linMag;
```
- What this code is doing is comparing the ActualLinearVelocity to the requested.  If the actual is less than requested it uses the positive and if greater uses the neg.
- Since this else that you are changing was returning when in reverse, my guess is ActualLinVel is not accounting for moving in reverse.  You would need to wrap an if and prob negate the ActualLinVel when in reverse gear, positive when not.  Make sense?

Mojojojooo, 2020-10-21:
Let me check once and get back if I face any issues. Thanks.

Mojojojooo, 2020-10-27:
Hi, @EricBoiseLGSVL thanks for the help we are now able to toggle between reverse and forward gear correctly. But now we want to access the status of that. Like if the REVERSE variable is true/ false. As the AD controller or wheel controller has no publisher part. Wanted to know how to publish those values and access in ROS.

EricBoiseLGSVL, 2020-10-27:
@Mojojojooo Great to hear it is working.  Now you will need to look at the ROS publish code just like you edited for the subscribe.  Edit so you can convert data that bridge can serialize and you are all set.