lgsvl/simulator #1063: I want to know the way of creating map that include HD map and mesh map.

kotakky, 2020-10-15:
I want to create a city map. The map include the mesh map like real world and HD map.
So, I want to know the data that can be treated in order to make that map.

I think point cloud map can be used in order to make those map.
Is it possible to make those map from open street map and HD map like apollo,lanelet2?

kotakky, 2020-10-15:
In  addition, I can get unity data (FBX format File) from Asset store.Is it possess to make the map that  include mesh map and  HD map  by annotating unity's data map?

Unity data map is like below image.

EricBoiseLGSVL, 2020-10-15:
@kotakky 
- We have a HDmap importer in the simulator [here](https://www.lgsvlsimulator.com/docs/map-annotation/#import-map-annotations)
- Point cloud can also be used instead of a mesh map but it does have limitations.  See [here](https://www.lgsvlsimulator.com/docs/pointcloud-import/)
- We are working on creating an OSM importer tool and should be released soon.  Yes you can import the HDMap see above
- Yes, you can import map into unity and use our HDmap tool to annotate manually.  You can even use San Francisco map as a guide for large city annotation since it is open source.

closed 2020-11-19 by kotakky

kotakky, 2020-11-19:
Point cloud can also be used instead of a mesh map but it does have limitations. 
→What do you think about that limitation?

EricBoiseLGSVL, 2020-11-19:
depends on what you need.  Testing perception or not