lgsvl/simulator #1065: ROS2, LGSVL, ros2-lgsvl-bridge, steering command control does not work as documented through lgsvl_msgs/VehicleControlData

jollysg, 2020-10-16:
Hey guys,

I recently installed the LGSVL simulator for using it with ROS2.0. I have been trying to control the ego vehicle in the simulator from a ROS2 node through ros2-lgsvl-bridge. The acceleration commands through the "acceleration_pct" member of lgsvl_msgs/VehicleControlData worked fine as documented. However, I saw an issue with the steering command. I am using "target_wheel_angle" member of the lgsvl_msgs/VehicleControlData for steering control. The ego doesn't respond to any value between  +0.69 and -0.69 radians. Above 0.69 (+ or -) the vehicle starts turning but the angle is already too large and the vehicle starts turning. The controls through keyboard work fine. Please check these video captures that depict this issue: 

1. Vehicle with ROS2 steering command -0.69 ([link](https://drive.google.com/file/d/1hO4ZuLYVpJ-B0Orn2Ax5yLT0bel-yJop/view?usp=sharing))
2. Vehicle with ROS2 steering command of -0.3 ([link](https://drive.google.com/file/d/11bwmPBTu-UT0TQpn4ZVpP6AF5M6AJkCv/view?usp=sharing))

Here are the details of the setup I am using:
LGSVL version: 2020.06
Map: San Francisco
Vehicle: Jaguar2015XE / Lexus2016RXHybrid with ROS2.0 bridge
ADE instructions: as per https://www.lgsvlsimulator.com/docs/autoware-auto-instructions/

In my ROS2.0 bridge, here is how I am publishing using the lgsvl_msgs/VehicleControlData message topic:

 ```
lgsvl_msgs::msg::VehicleControlData control_data;
 control_data.acceleration_pct = 0.25;
 control_data.target_wheel_angle = -0.69; // in radians
 control_pub_->publish(control_data);
```
Please let me know if you feel I am missing any configurations or this is a bug. Also, I am willing to provide any other information you need to get to this.

Thanks,
Jasprit

EricBoiseLGSVL, 2020-10-16:
Thanks for the detailed post.  Any idea @hadiTab ?

jollysg, 2020-10-16:
Also tried sending both steering angle as well as steering rate command, but no difference. Attached is a screenshot of ros2 topic echo for the topic. 

[image]

Also, for the reference, here is the JSON script for the Lexus vehicle I am using, in case that helps. The Jaguar I tried with also had a similar one. The scripts are largely from the instructions pages on the LGSVL documentation.

```
[
  {
    "type": "Transform",
    "name": "base_link",
    "transform": {
      "x": -0.015,
      "y": 0.369,
      "z": -1.37,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "CAN-Bus",
    "name": "CAN Bus",
    "params": {
      "Frequency": 10,
      "Topic": "/lgsvl/state_report"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS Device",
    "name": "GPS",
    "params": {
      "Frequency": 12.5,
      "Topic": "/gnss/fix",
      "Frame": "gnss",
      "IgnoreMapOrigin": true
    },
    "parent": "base_link",
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS Odometry",
    "name": "GPS Odometry",
    "params": {
      "Frequency": 30.0,
      "Topic": "/lgsvl/gnss_odom",
      "Frame": "odom",
      "ChildFrame": "base_link",
      "IgnoreMapOrigin": true
    },
    "parent": "base_link",
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "IMU",
    "name": "IMU",
    "params": {
      "Topic": "/imu/imu_raw",
      "Frame": "imu"
    },
    "parent": "base_link",
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Lidar",
    "name": "LidarFront",
    "params": {
      "LaserCount": 16,
      "MinDistance": 2.0,
      "MaxDistance": 100,
      "RotationFrequency": 10,
      "MeasurementsPerRotation": 360,
      "FieldOfView": 20,
      "CenterAngle": 0,
      "Compensated": true,
      "PointColor": "#ff000000",
      "Topic": "/lidar_front/points_raw",
      "Frame": "lidar_front"
    },
    "parent": "base_link",
    "transform": {
      "x": 0.022,
      "y": 1.49,
      "z": 1.498,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Lidar",
    "name": "LidarRear",
    "params": {
      "LaserCount": 16,
      "MinDistance": 2.0,
      "MaxDistance": 100,
      "RotationFrequency": 10,
      "MeasurementsPerRotation": 360,
      "FieldOfView": 20,
      "CenterAngle": 0,
      "Compensated": true,
      "PointColor": "#ff000000",
      "Topic": "/lidar_rear/points_raw",
      "Frame": "lidar_rear"
    },
    "parent": "base_link",
    "transform": {
      "x": 0.022,
      "y": 1.49,
      "z": 0.308,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "3D Ground Truth",
    "name": "3D Ground Truth",
    "params": {
      "Frequency": 10,
      "Topic": "/simulator/ground_truth/threed_detections"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "3D Ground Truth Visualizer",
    "name": "3D Ground Truth Visualizer",
    "params": {
      "Topic": "/simulator/ground_truth/threed_visualize"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "2D Ground Truth",
    "name": "2D Ground Truth",
    "params": {
      "Frequency": 10,
      "DetectionRange": 100,
      "Topic": "/simulator/ground_truth/twod_detections"
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "2D Ground Truth Visualizer",
    "name": "2D Ground Truth Visualizer",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 2000,
      "Topic": "/simulator/ground_truth/twod_visualize"
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Vehicle Control",
    "name": "Autoware Car Control",
    "params": {
      "Topic": "/lgsvl/vehicle_control_cmd"
    }
  },
  {
    "type": "Vehicle State",
    "name": "Autoware Auto Vehicle State",
    "params": {
      "Topic": "/lgsvl/vehicle_state_cmd"
    }
  },
  {
    "type": "Keyboard Control",
    "name": "Keyboard Car Control"
  },
  {
    "type": "Clock",
    "name": "Simulation Clock",
    "params": {
      "Topic": "/lgsvl/clock"
    }
  }
]

```

hadiTab, 2020-10-16:
@jollysg I just tested it on my end and I'm seeing similar behavior. It seems like it's a bug. We will look into it more and fix it. Thanks for reporting.

jollysg, 2020-10-16:
@hadiTab Is there an existing work around that I can use through ROS2 in the meanwhile for controlling the ego vehicle?

hadiTab, 2020-10-16:
@jollysg you can download the source code, fix it and either run the simulator in the editor or build yourself a new binary and use that. Make sure to use the `release-2020.06` branch and **not** master.

To fix the bug paste the following code [over here](https://github.com/lgsvl/simulator/blob/release-2020.06/Assets/Scripts/Bridge/Ros2/Ros2Conversions.cs#L380)
```csharp
            else
            {
                wheelAngle = data.target_wheel_angle;
            }
```

jollysg, 2020-10-17:
Thanks @hadiTab I will try the steps and get back to you if I have questions.

jollysg, 2020-10-19:
@hadiTab I was able to compile the release-2020.06 branch source code with the fix you mentioned and I am able to control the vehicle through ROS2 lgsvl_msgs now. Thanks for the quick turn around.

closed 2020-10-20 by EricBoiseLGSVL