lgsvl/simulator #1066: Is it possible to add camera sensor on top of traffic lights

PS06, 2020-10-16:


EricBoiseLGSVL, 2020-10-16:
No, but you can change source code to do this.  You can also look at Controllables for this

soldierofhell, 2020-12-11:
Hi @EricBoiseLGSVL, I'd also like to have the possibility to attach camera to some static object like traffic lights, building or bridge. Camera Sensor is like natural candidate but the problem is the sensors are intended for ego vehicle only. Could you guide me how to approach to such code change so that we allow Controllables to have sensors (preferably with shared sensors code withe ego vehicle). Regards,

EricBoiseLGSVL, 2020-12-11:
We are looking into extending sensors to static objects in the environment.  The framework is not set for this now and we will hopefully have a solution soon.

soldierofhell, 2020-12-11:
Thanks, let us now if you need some testing, etc.