lgsvl/simulator #1067: Creating Digital Twin with pointcloud map

praveenputhettu, 2020-10-16:
What all things needs to considered for creating a digital twin map form a LiDAR pointcloud map?

EricBoiseLGSVL, 2020-10-16:
@praveenputhettu What do you mean "all things considered"?  We have documentation on importing pointcloud maps [here](https://www.lgsvlsimulator.com/docs/pointcloud-import/) and importing HDmaps [here](https://www.lgsvlsimulator.com/docs/map-annotation/#import-map-annotations).  Is something not working or are you having a specific issue?

praveenputhettu, 2020-10-19:
Thank you @EricBoiseLGSVL for the response,
Sorry for the delay in my reply.
The digital twin of GoMentum Station section Urban A looks great(https://www.lgsvlsimulator.com/2020/01/21/2019-12-release/)], in creating such quality digital twin what all things i need to consider?
Is it possible to create such quality Digital Twin with Lidar point clouds alone? If not what else i need to consider?
should i need to do any post processing on point cloud before or after importing to increase the quality to create a digital twin?

EricBoiseLGSVL, 2020-10-19:
We have a service to take point cloud data and process it automatically to achieve such a high quality render (it's even better now :) ).  We have a certain criteria for collecting the data for this service, I'll link some people here to discuss with you.   @daviduhm @zelenkovsky  @lemketron

praveenputhettu, 2020-10-21:
@daviduhm @zelenkovsky @lemketron, Can i know more about the service?

daviduhm, 2020-10-28:
@praveenputhettu For creating Digital Twin maps like GoMentum Station, you'll need to collect sensor data in Rosbag including Lidar point clouds, camera images, and GPS/IMU data. All sensors must be calibrated and required to have calibration data as well. From there, there are post processing steps such as localization, point cloud colorization, or point cloud filtering in order to make the final colorized point cloud data out of Rosbag that you can import it into the simulator.

We have a service to take sensor data from customers and create Digital Twin maps, but currently it's in beta and only available for our beta customers. If you're interested and would like to know more in detail, please contact us via email at contact@lgsvlsimulator.com.