lgsvl/simulator #1068: How to use g29 steering wheel on windows

pdcxs, 2020-10-16:
I buy a g29 steering wheel, but the simulator cannot detect it. I'm on windows platform, the drivers is properly installed. I have tried rebuild the simulator from source codes, but the vehicle still cannot be controlled by the g29 steering wheel.

Is there any simple method to enable g29 controller?

EricBoiseLGSVL, 2020-10-16:
did you add the WheelControlSensor [here](https://www.lgsvlsimulator.com/docs/sensor-json-options/#wheel-control)?

pdcxs, 2020-10-16:
> did you add the WheelControlSensor [here](https://www.lgsvlsimulator.com/docs/sensor-json-options/#wheel-control)?

Yes, I've already added WheelControlSensor in the vehicle sensor settings.

EricBoiseLGSVL, 2020-10-16:
@pdcxs I have seen Unity not be able to recognize the wheel, are you able to open the Editor and then the InputSystem debug.  You should see the wheel in this menu.  If not you may still have driver issues
[image]

pdcxs, 2020-10-17:
> @pdcxs I have seen Unity not be able to recognize the wheel, are you able to open the Editor and then the InputSystem debug. You should see the wheel in this menu. If not you may still have driver issues
> [image]

Yes, in the Unity editor, the steering wheel is in the device list. But still, the vehicle in the simulator has no response for the steering wheel input.

EricBoiseLGSVL, 2020-10-19:
- Double click the wheel in the Input Debugger and see if the inputs are being received
- If so, screen shot the vehicle with the wheel control sensor attached
- Check the console panel to see if there are any errors and post them if so
- We will need more info to debug this, please add debug logs to capture input in Wheel control sensor and post the player.log to see if inputs are being received at runtime.

pdcxs, 2020-10-20:
> * Double click the wheel in the Input Debugger and see if the inputs are being received
> * If so, screen shot the vehicle with the wheel control sensor attached
> * Check the console panel to see if [...]

I double click the wheel in the input debugger, but nothing has happened. I try to double click the keyboard and mouse, a little window will pop up, but the wheel not show this window. So, is this the problem of the driver?

pdcxs, 2020-10-20:
I find when I double click the wheel, it reports that "NotImplementedException: Reading int stradding int boundary"
[image]

EricBoiseLGSVL, 2020-10-20:
Yes, this seems like an issue with the driver.  Do you have the logitec software installed, not just drivers?  If not I would remove drivers and just use software to install correct drivers

pdcxs, 2020-10-21:
I'm tired of trying fix this problem. I just write a little code, first read the inputs of the G29 with pygame_sdl2 and then use the inputs to control the vehicles. Thanks for your answers any way.

closed 2020-10-21 by pdcxs