lgsvl/simulator #1069: the on_collision callback is not invoked

yl2280, 2020-10-16:
I am trying to use the on_collision callback on ego vehicle, but I found out when the collision happens, the callback is not invoked. I am using apollo 3.5 + lgsvl2019.09 with the map of BorregasAve. It is confusing, since the collision callback is working in lgsvl2019.05 with the san francisco map.  I did install the pythonApi compatible with lgsvl2019-09. Could someone help me with this.

EricBoiseLGSVL, 2020-10-16:
@luqiang21 any idea?

luqiang21, 2020-10-17:
@EricBoiseLGSVL I didn't try the callback for a long time. @yl2280 Can I ask why are you using such an old release which was released one year ago? Did you try the latest release 2020.06?

EricBoiseLGSVL, 2020-10-19:
Yes, @luqiang21 is right.  Please update to the latest release.  Note: it is a branch, not master

EricBoiseLGSVL, 2020-10-20:
Also be sure that your API installation is in developer mode because if it isn't, when you pull latest changes, it will not update.  You will then need to reinstall it.  See [here](https://www.lgsvlsimulator.com/docs/python-api/#quickstart) and the example has `-e`

huongvn, 2020-12-29:
I wite my own code to detect collision , but not work
when 2 car collision
here my code:
`#!/usr/bin/env python3
#
# Copyright (c) 2019-2020 LG Electronics, Inc.
#
# This software contains code licensed as described in LICENSE.
#

import os
from environs import Env
import lgsvl
import time

env = Env()

sim = lgsvl.Simulator(env.str("LGSVL__SIMULATOR_HOST", "127.0.0.1"), env.int("LGSVL__SIMULATOR_PORT", 8181))
if sim.current_scene == "SingleLaneRoad":  #map
    sim.reset()
else:
    sim.load("SingleLaneRoad")  #map

spawns = sim.get_spawn()

# ego vehicle
state = lgsvl.AgentState()
state.transform = spawns[0]
ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", "Lincoln2017MKZ (Apollo 5.0)"), lgsvl.AgentType.EGO, state)
# ego start location
forward = lgsvl.utils.transform_to_forward(spawns[0])
state.transform.position += 0 * forward  # 0m forwards
print("DEBUG: on here 0")
# school bus, perpendicular to road, stopped
state = lgsvl.AgentState()
state.transform.position = spawns[0].position + 40.0 * forward   #40m forwards --start location
state.transform.rotation.y = spawns[0].rotation.y + 90.0         #90 degree form defaul (y)
bus = sim.add_agent("SchoolBus", lgsvl.AgentType.NPC, state)

print("DEBUG: on here 1")

def on_collision(agent1, agent2, contact):
  name1 = "STATIC OBSTACLE" if agent1 is None else agent1.name
  name2 = "STATIC OBSTACLE" if agent2 is None else agent2.name
  print("{} collided with {} at {}".format(name1, name2, contact))

print("DEBUG: on here 2")

ego.on_collision(on_collision)

print("DEBUG: on here 3")

# start 
input("Press Enter to run")

print("DEBUG: on here 4")
# car automation control
# VehicleControl objects can only be applied to EGO vehicles
# You can set the steering (-1 ... 1), throttle and braking (0 ... 1), handbrake and reverse (bool)
c = lgsvl.VehicleControl()
c.throttle = 0.3
c.steering = 0.0
print("DEBUG: on here 5")
# a True in apply_control means the control will be continuously applied ("sticky"). False means the control will be applied for 1 frame
ego.apply_control(c, True)
print("DEBUG: on here 6")
# run 1 minute
sim.run(time_limit = 20.0) 
print("DEBUG: on here 7")
`

huongvn, 2020-12-29:
> I wite my own code to detect collision , but not work
> when 2 car collision
> here my code:
>
> ```
> #!/usr/bin/env python3
>
> #
> # Copyright (c) 2019-2020 LG Electronics, Inc.
> #
> # This software contains code [...]

@EricBoiseLGSVL