lgsvl/simulator #1070: Fisheye camera calibration issue

asifsid-32, 2020-10-19:
Hello LGSVL team, 
We are trying to calibrate the fisheye camera sensors present in the available sensors of lgsvl simulator. We tried correcting the fisheye image using the given distortion coefficients ->  [-0.25349, 0.11868, 0, 0] and
 camera matrix -> 
                                   [ 772, 0, 640,
                                     0, 772, 360,
                                     0, 0, 1]
given by simulator team in given issue https://github.com/lgsvl/simulator/issues/1047.

Below attached is the fisheye corrected image which we are getting after undistorting the image which is not properly fisheye corrected as we can see the image is still bulgy in the corners.

Input Image -> 
[image]

Output Fisheye corrected image -> 
[image]

Please let me know what can be done to get the correct fisheye image, it seems that the distortion coefficients and camera matrix are not correct.

EricBoiseLGSVL, 2020-10-19:
@asifsid-32 @rongguodong has given you the camera matrix.  What is the issue with the matrix?  Are you calculating the horizontal fov then?  The previous issue was answered with a vertical fov of 50, is this not correct?

asifsid-32, 2020-10-19:
> @asifsid-32 @rongguodong has given you the camera matrix.  What is the issue with the matrix?  Are you calculating the horizontal fov then?  The previous issue was answered with a vertical fov of 50, is this not correct?

The matrix given is not giving the correct fisheye corrected output. 

Also we needed horizontal FOV which is not available in the documentation. Is it possible that you can provide us with the horizontal FOV ?

EricBoiseLGSVL, 2020-10-19:
@asifsid-32 Unfortunately we cannot help you on this.  You will need to convert the matrix given to the horizontal fov you need.  I think @rongguodong has given you a great amount of help in #1047 and this is a duplicate of that issue.  Sorry we cannot be more help but our limited engineering time can't solve this for you.  Maybe someone in the community knows how to solve this.