lgsvl/simulator #1072: New ego vehicle is stuck and doesn't move during simulation.

nike-96, 2020-10-19:
Hi, I am running on 2020.06 release and I have just created a new ego vehicle starting from the Jaguar2015XE reference model, following all the steps (Unity 2019.3.15f1). 
As I start the simulation though, the vehicle can't move, neither with keyboard control nor with G920 Steering wheel (I have added both the sensors to the vehicle). All components and features seems to be exactly the same as the Jaguar, but still I have this problem. 

PS: I have checked and with other vehicles I have no problems.

EricBoiseLGSVL, 2020-10-19:
You most likely do not have the axles set right in the VehicleSMI.cs, please check this.
You also may have moved or edited the wheels and their pivots are not correct.  Please highlight these elements in the scene view in the editor and post screen shots so we can see.

nike-96, 2020-10-20:
[image]
[image]
[image]
[image]
[image]

Here they are, the axles of VehicleSMI.cs and all the wheels. I still can't find the issue, I hope you can help me.

EricBoiseLGSVL, 2020-10-20:
Please highlight these elements in the scene view in the editor and post screen shots so we can see like so
[image]
[image]
[image]
[image]

Another thing, be sure to tag the top object Player and the Layer and all children Agent like so
[image]

nike-96, 2020-10-21:
Here you go 
[image]
[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2020-10-21:
This is a problem.  Why is the center pivot so far off?
[image]

Please look at how the original ego is created.  You can drag it in scene to compare to the one you created.
[image]
[image]
[image]

You need meshes and holder object for wheel colliders zero'd to parent.  Please check this for body and wheel collider holder.

nike-96, 2020-10-21:
I did check the meshes but I didn't find anything strange. If you look at the center pivot of the Jaguar2015XE.prefab, it is in the same position as the mine one. I didn't understand your suggestion probably.

EricBoiseLGSVL, 2020-10-21:
From your image, the vehicle mesh is not aligned at the parents origin.  You can see the gizmo.  This makes me think other children are not aligned.  Please each gameobject in the hierarchy to make sure they are at origin.  (vehicle mesh and wheel collider holder)

nike-96, 2020-10-22:
[image]

This is the original ego vehicle of the Jaguar, I just re-cloned it from git-hub to be sure but as you can see the pivot is in a different position from yours one. Where did you take the vehicle model? From https://github.com/lgsvl/Jaguar2015XE/tree/master ?

nike-96, 2020-10-22:
I have found the issue. I had followed the step 14 of https://github.com/lgsvl/simulator/issues/401, but not for the root of the prefab. Indeed, it was empty. After properly changing it, it worked. Thank you so much for the support.

EricBoiseLGSVL, 2020-10-22:
No problem, glad to help.  FYI we have a youtube video [here](https://youtu.be/R_Z65kimCII) for steps and we will be updating our docs on this very soon.

closed 2020-10-22 by EricBoiseLGSVL