lgsvl/simulator #1074: How to do the "Render Pipeline" if there is no .mat files in the custom assets

qiongwu1003, 2020-10-20:
System : Win10
Unity version : 2019.3.15f1
Simulator code version : 8b02c852

I export a map scene from RoadRunner, the exports includes a Filmbox (.fbx) file containing the 3D objects in a scene along with an additional XML file to hold extra data for materials and traffic signals in the scene.
On the Unity side, a set of scripts are included in the RoadRunnerUnityTool asset package to help import the FBX® file using the information stored in the XML file. The script handles the following details:
 Setting up materials
 Adding colliders to roads and terrain
 Setting up the components of traffic signals

The scene imported using the RoadRunnerUnityTool plugin and works fine in a normal Unity project, but when imported into the simulator project they become pink. 

I saw it needs to do "Render Pipeline" thing in [here](https://github.com/lgsvl/simulator/issues/666).  But there is no .mat files in my scene assets. Is there another way to do that?

I am attaching pictures to help explain the situation better.

In the normal Unity project:
[image]

In the simulator project:
[image]

Ps : when I search "All Materials" I can see bellow : 
[image]

Any tips or help on this would be great, Thanks!

EricBoiseLGSVL, 2020-10-20:
Roadrunner does not support HDRP shaders and this is why the materials are pink.  If you use roadrunner, you will need to the conversion manually and it is not easy.  We also do not support the other RoadRunner features, e.g., custom signal lights
- Export imported meshes materials here to a Materials folder right next to mesh
[image]

- Then convert materials to HDRP like so and reassign Textures and values.  Roadrunner does not support HDRP and shaders are very different.  All these material parameters will need to be adjusted.
[image]

- Again, RoadRunner is using old Unity render pipeline and is not supporting HDRP, you will need to manually make the conversions.  Also, RR uses multiple meshes and materials with lots of alpha for roads, lines, oil marks, etc.  This is not performant for game engines, causes issues with alpha clipping and has many issues using LGSVL Simulator features.  We expect one mesh with one material for road surfaces.  Then, we use Unity's new shader graph shader to apply wetness, puddles, intensity and damage onto this one mesh.  This is more performant and looks more realistic.

- I recommend cleaning up the mesh in Maya or Blender and setting default Unity shaders before importing in Simulator projects.  This gives you more control and you can cleanup the multiple meshes and materials.
- If this is not possible, then I recommend using our HDMap importer that then can convert the map to a mesh.  Then you can add decoration to the environment.

qiongwu1003, 2020-10-21:
Thanks for your reply.

And I tried the HDMap importer way.

step1: Export OpenDRIVE(.xodr) file from RoadRunner.
step2: Import the OpenDRIVE file to simulator. And I referenced [this](https://www.youtube.com/watch?v=bZk5kY-YgEM) vedio.
           There is nothing but only the MapSignal in the scene, and with errors. Does that mean the RR OpenDRIVE file can not be used in the simulator?
[image]

Plus, in my simulator there is no "Build HD Map Mesh" item, how do i fix it?
[image]

Thanks.

luqiang21, 2020-10-21:
@qiongwu1003 I see you are using code from `master` branch which is not the latest release. You should use branch `release-2020.06` for the most recent release. You should be able to find `Build HD Map Mesh` button in the simulator.

qiongwu1003, 2020-10-22:
@luqiang21 Thanks your reply, I will try the release-2020.06 code. And about the "HDMap importer" question, any suggestion?

luqiang21, 2020-10-22:
@qiongwu1003 We have several fixes for the HD map importer, you can try it as well. Let me know if you still have problems.

qiongwu1003, 2020-10-22:
@luqiang21 
Hi, I tried the release-2020.06 code,  and I imported the map success. But I still can  not find the "Build HD Map Mesh" item.
And i also checked the "simulator\Assets\Scripts\Editor" path, there is no file about "Build HD Map Mesh" item. 

[image]

Thanks.

luqiang21, 2020-10-22:
@qiongwu1003 That's great to hear you can import the HD map. Sorry I was mistaken that as you can in the title of the [video](https://www.youtube.com/watch?v=bZk5kY-YgEM), Build Mesh from HDMap is one of premium features which you need to contact our team (contact@lgsvlsimulator.com) for more details if you want to try or use it.

qiongwu1003, 2020-10-22:
@luqiang21 Thank you very much. And there is no more question.

closed 2020-10-22 by qiongwu1003