lgsvl/simulator #1076: LGSVL does not start on Ubuntu 20.04

kmitri-dxc, 2020-10-20:
When starting the pre-compiled version downloaded from the release 2020.06, I run the simulator, a black screen appears and then goes away.

***
System:    Host: kmitri-Alienware-Area-51m Kernel: 5.4.0-51-generic x86_64 bits: 64 Desktop: Gnome 3.36.4 
           Distro: Ubuntu 20.04.1 LTS (Focal Fossa) 
Machine:   Type: Laptop System: Alienware product: Alienware Area-51m v: 1.12.0 serial: <superuser/root required> 
           Mobo: Alienware model: Alienware Area-51m v: A00 serial: <superuser/root required> UEFI: Alienware v: 1.12.0 
           date: 07/07/2020 
CPU:       Topology: 8-Core model: Intel Core i9-9900K bits: 64 type: MT MCP L2 cache: 16.0 MiB 
           Speed: 800 MHz min/max: 800/5000 MHz Core speeds (MHz): 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800 7: 800 8: 800 
           9: 800 10: 800 11: 800 12: 800 13: 800 14: 800 15: 800 16: 800 
Graphics:  Device-1: Intel UHD Graphics 630 driver: i915 v: kernel 
           Device-2: NVIDIA TU104BM [GeForce RTX 2080 Mobile] driver: nvidia v: 450.80.02 
           Display: x11 server: X.Org 1.20.8 driver: modesetting,nvidia unloaded: fbdev,nouveau,vesa 
           resolution: 1920x1080~144Hz 
           OpenGL: renderer: GeForce RTX 2080/PCIe/SSE2 v: 4.6.0 NVIDIA 450.80.02 
Network:   Device-1: Realtek driver: r8169 
           IF: enp69s0 state: down mac: 98:e7:43:97:32:67 
           Device-2: Intel Wireless-AC 9260 driver: iwlwifi 
           IF: wlp70s0 state: up mac: 28:7f:cf:a5:3d:59 
           IF-ID-1: docker0 state: down mac: 02:42:ab:af:ac:06 
Drives:    Local Storage: total: 1.38 TiB used: 50.31 GiB (3.6%) 
           ID-1: /dev/nvme0n1 vendor: SK Hynix model: PC601A NVMe 512GB size: 476.94 GiB 
           ID-2: /dev/nvme1n1 vendor: SK Hynix model: PC601A NVMe 512GB size: 476.94 GiB 
           ID-3: /dev/sda vendor: Seagate model: ST1000LX015-1U7172 size: 931.51 GiB 

Partition: ID-1: / size: 467.96 GiB used: 50.30 GiB (10.7%) fs: ext4 dev: /dev/nvme0n1p2 

P.S: Nvidia driver installed from the recommended ppa.

******

**Player LOG:**

kmitri@kmitri-Alienware-Area-51m:~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator$ cat Player.log
Mono path[0] = '/home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
Mono config path = '/home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Loading player data from /home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
Desktop is 1920 x 1080 @ 144 Hz
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
Vulkan detection: 2
Initialize engine version: 2019.3.15f1 (59ff3e03856d)
[Subsystems] Discovering subsystems at path /home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[... 122 lines cut ...]
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Controllables/controllable_TrafficCone/manifest 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library /home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/data-0x3947490.so
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Sensors/sensor_CheckControlSensor/manifest 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library /home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/data-0x3948ca0.so
found manifest at Sensors/sensor_VelodyneLidarSensor/manifest 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library /home/kmitri/Simulators/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/data-0x38f8700.so
Loaded 3 NPCs behaviours and 6 NPC models in 00:00:00.1238371 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x460000f
  Serial number of failed request:  265
  Current serial number in output stream:  271

EricBoiseLGSVL, 2020-10-20:
Thanks for the debug logs, it is very helpful.  Did you see this?
```
Vulkan API version 1.1.0 (1.2.131 supported by driver)
Vulkan vendor=[Intel] id=[8086]
Vulkan renderer=[Intel(R) UHD Graphics 630 (CFL GT2)] id=[3e98]
Vulkan device type 1
```
This looks like you are running the application with the Intel, not the nvidia GPU.  Please try to force the application to use the GPU.  One way is in the Nvidia control panel, force LGSVL Simulator app to use GPU not Intel.  Hope this helps

mjshiggins, 2020-10-20:
@EricBoiseLGSVL Oddly enough, I just started experiencing this today. I'm not sure what could have changed to start causing this across different users, but what is the standard way to force the Nvidia GPU (I'm personally using LGSVL inside a Docker container)?

EricBoiseLGSVL, 2020-10-20:
Not sure on Linux, I use Windows ;) @lemketron or @zelenkovsky

kmitri-dxc, 2020-10-20:
Thanks @EricBoiseLGSVL, I couldn't figure out how to force it to use the NVIDIA GPU, but It worked well in a docker container using NVIDIA Container Toolkit. 

 @mjshiggins  Install Nvidia Container Toolkit and run docker with --gpu all.

closed 2020-10-20 by kmitri-dxc

shr-project, 2020-10-20:
It works for me in Ubuntu 20.04 with docker, but I have just 1 GPU in the system:
```
$ nvidia-container-cli info
NVRM version:   450.80.02
CUDA version:   11.0

Device Index:   0
Device Minor:   0
Model:          GeForce GTX 1060 6GB
Brand:          GeForce
GPU UUID:       GPU-5fdec0a6-ada0-70ed-c6b4-bce14668482f
Bus Location:   00000000:01:00.0
Architecture:   6.1
```

Maybe you can restrict NVIDIA_VISIBLE_DEVICES to include the id of your nvidia card, so that it doesn't expose the intel one in docker?