lgsvl/simulator #1077: problem with using gps coordinate in Autoware

momala, 2020-10-20:
Hi,
I want to use GNSS for better localizing. I attached my GPS frame on my base-link frame inside the simulator. while I move my vehicle, the GPS frame rotates along its axis too much (based on Rviz representation). And it does not let me to localized based on GPS data. Honestly, I can't understand what's going on that cause the GPS frame to rotate.
[image]
here is my GPS frame when my vehicle starts (map: BorregasAve and position: Spawn[1]). as you can see the z axis looks downward!!
[image]
this problem does not let me localize as you can see.
do you have any Idea?

EricBoiseLGSVL, 2020-10-20:
@hadiTab any ideas?  @momala Can you please post your sensor configuration so we can see how it is parenting the sensors?

momala, 2020-10-21:
> @hadiTab any ideas? @momala Can you please post your sensor configuration so we can see how it is parenting the sensors?

`[
    {   
        "type": "GPS Device", 
        "name": "GPS",  
        "params": {
            "Frequency": 10,
            "Topic": "/nmea_sentence",
            "Frame": "gps",
            "IgnoreMapOrigin": true
        },

        "transform": {
            "x": 0, 
            "y": 0, 
            "z": 0, 
            "pitch": 0, 
            "yaw": 0, 
            "roll": 0
        }
    },

    {   
        "type": "GPS Odometry", 
        "name": "GPS Odometry",
        "params": {
            "Frequency": 12.5, 
            "Topic": "/odom", 
            "Frame": "gps", 
            "IgnoreMapOrigin": true
        },

        "transform": {
            "x": 0, 
            "y": 0, 
            "z": 0, 
            "pitch": 0, 
            "yaw": 0, 
            "roll": 0
        }
    },

    {   
        "type": "IMU", 
        "name": "IMU",
        "params": {
            "Topic": "/imu_raw", 
            "Frame": "imu"
        },

        "transform": {
            "x": 0, 
            "y": 0, 
            "z": 0, 
            "pitch": 0, 
            "yaw": 0, 
            "roll": 0
        }
    },

    {   
        "type": "Lidar",
        "name": "VLP_Aft",
        "params": {
            "LaserCount": 16,
            "FieldOfView": 30,
            "CenterAngle": 0,
            "MinDistance": 0.10,
            "MaxDistance": 100,
            "RotationFrequency": 10,
            "MeasurementsPerRotation": 360,
            "Compensated": true,
            "PointColor": "#ff000000",
            "PointSize": 3,
            "Topic": "/points_aft", 
            "Frame": "velodyne_a"
        },

        "transform": {
            "x": 0,
            "y": 2.25,
            "z": 0,
            "pitch": -10,
            "yaw": 0,
            "roll": 0
        }
    },

    {
        "type": "Lidar",
        "name": "VLP_Fore",
        "params": {
            "LaserCount": 16,
            "FieldOfView": 30,
            "CenterAngle": 0,
            "MinDistance": 0.10,
            "MaxDistance": 100,
            "RotationFrequency": 10,
            "MeasurementsPerRotation": 360,
            "Compensated": true,
            "PointColor": "#0800ff",
            "PointSize": 3,
            "Topic": "/points_fore", 
            "Frame": "velodyne_f"
        },

        "transform": {
            "x": 0,
            "y": 2.25,
            "z": 2.6,
            "pitch": 10,
            "yaw": 0,
            "roll": 0
        }
    },

    {
        "type": "Lidar",
        "name": "Lidar_left",
        "params": {
            "LaserCount": 16,
            "FieldOfView": 85,
            "CenterAngle": -45,
            "MinDistance": 0.20,
            "MaxDistance": 100,
            "RotationFrequency": 10,
            "MeasurementsPerRotation": 360,
            "Compensated": true,
            "PointColor": "#29ff1a",
            "PointSize": 3,
            "Topic": "/points_left",
            "Frame": "velodyne_l"
        },

        "transform": {
            "x": -0.7,
            "y": 1.5,
            "z": 2.455,
            "pitch": 0,
            "yaw": 0,
            "roll": 90
        }
    },

    {
        "type": "Lidar",
        "name": "Lidar_right",
        "params": {
            "LaserCount": 16,
            "FieldOfView": 85,
            "CenterAngle": -45,
            "MinDistance": 0.20,
            "MaxDistance": 100,
            "RotationFrequency": 10,
            "MeasurementsPerRotation": 360,
            "Compensated": true,
            "PointColor": "#ffff2e",
            "PointSize": 3,
            "Topic": "/points_right",
            "Frame": "velodyne_r"
        },

        "transform": {
            "x": 0.7,
            "y": 1.5,
            "z": 2.455,
            "pitch": 0,
            "yaw": 0,
            "roll": -90
        }
    },

    {
        "type": "Keyboard Control", 
        "name": "Keyboard Car Control"
    },
    {
        "type": "Wheel Control", 
        "name": "Wheel Car Control"
    },
    {
        "type": "Vehicle Control", 
        "name": "Autoware Car Control",
        "params": {
            "Topic": "/vehicle_cmd"
        }
    }
]`
I add four lidar sensors which are attached to their specific TF related to the base-link. when I used the lidar published data I figure out that the rotation is not the same as I defined inside the LGSVL.[forexample the side lidars point cloud is upward and need 90 degree rotation] here is my static TF representation in my launch file:
`<node pkg="tf"  type="static_transform_publisher" name="velodyne_to_vlodyne_a" args="0 0 0 0 -0.174 0 /velodyne /velodyne_a 10" />
<node pkg="tf"  type="static_transform_publisher" name="velodyne_to_vlodyne_f" args="0 0 0 0 0.174 0 /velodyne /velodyne_f 10" />
<node pkg="tf"  type="static_transform_publisher" name="velodyne_to_vlodyne_l" args="0 0.7 -0.75 0 0 -1.57 /velodyne /velodyne_l 10" />
<node pkg="tf"  type="static_transform_publisher" name="velodyne_to_vlodyne_r" args="0 -0.7 -0.75 0 0 1.57 /velodyne /velodyne_r 10" />`

But my main problem now is the GPS frame. I attached it exactly on base_link (inside lgsvl). not only it is not on its position at the start point, but also when I move the vehicle the GPS rotation changes!!

hadiTab, 2020-10-21:
@momala I'm not completely understanding the rotation problem you are mentioning with the lidars, but keep in mind that the coorindates you enter in the JSON file are unity coordinates (which is left-handed ZXY - with Z forward and Y pointing up), but ROS uses a right-handed XYZ coordinates system. So make sure you do the conversion between the two.

As for the difficulty you are having with the localization, this seems to be an issue on the Autoware side. What steps are you going through to run the localization? How are you setting it up? Also, what version of Autoware are you using. GPS localization only works in 1.14.

momala, 2020-10-21:
> @momala I'm not completely understanding the rotation problem you are mentioning with the lidars, but keep in mind that the coorindates you enter in the JSON file are unity coordinates (which is [...]

Thank you @hadiTab for your response and sorry for my bad explanation. I thought the JSON TF is right hand but with a different rotation. So it needs some modification. As I have 4 lidars I should map them in their correct position(otherwise it creates some localization errors). But my problem for the GPS is when I use the nmea2tfpose, it creates the GPS frame that is not rotated correctly. the positioning is correct but the rotation is not. do you think it is also because of the differences with right and left-hand coordinates?

For the localization:
I use both nmea2tfpose and ndt matching.
`  
  <include file="$(find gnss_localizer)/launch/nmea2tfpose.launch">
    <arg name="plane" value="0" />
  </include> 

  
  <include file="$(find lidar_localizer)/launch/ndt_matching.launch">
    <arg name="get_height" value="$(arg get_height)" />
  </include>`

My Autoware version is 1.14.
So first I want to try to check the TF again.

momala, 2020-10-21:
> @momala I'm not completely understanding the rotation problem you are mentioning with the lidars, but keep in mind that the coorindates you enter in the JSON file are unity coordinates (which is [...]

@hadiTab , I should mention that when I rotate the vehicle inside the simulator the GPS frame which is based on the nmea2tfpose, start to rotate. the positioning is correct but the rotation is changing. do you think it is totally related to the Autoware conversion or maybe it can be related to the LGSVL?
[image]
[image]
[image]
these are the picture of different rotation related to the base-link.
I am trying to tweak the nmea2tfpose to see if I can manage it.

zerolover, 2021-02-07:
@hadiTab There are two question
First is duplicated conversion.
GPS sensor has converted orientation from left hand to right
https://github.com/lgsvl/simulator/blob/dc173a2df5cdff155455074d49529cccb58ef3f3/Assets/Scripts/Sensors/GpsSensor.cs#L133
But when publish, lgsvl converted it again, see
https://github.com/lgsvl/simulator/blob/dc173a2df5cdff155455074d49529cccb58ef3f3/Assets/Scripts/Bridge/Ros/RosWriter.cs#L200

Second is wrong Euler rotation. In Unity the Euler angle is ZXY, so when calling `message.Orientation.eulerAngles;`, you  will get euler in ZXY(Tait–Bryan angles), not ZYX(Tait–Bryan angles) in ros.
see https://www.ros.org/reps/rep-0103.html

JM998, 2021-02-13:
My question isn't similar to those above. I have the same problem that someone has commented in GPS data error#657, while it is said that this problem has been fixed in the latest release.
I will be much appreciate if you can answer my question that the GPS data in simulator is stiil different from the one in the nmea_sentence msgs, even though the version of simulator is latest, 2020.06 for windows.

JM998, 2021-03-04:
@hadiTab @EricBoiseLGSVL @momala 
May I have a question? My question isn't similar to those above. I have the same problem that someone has commented in GPS data error #657, while it is said that this problem has been fixed in the latest release.
I will be much appreciate if you can answer my question that the GPS data in simulator is still different from the one in the nmea_sentence msgs, even though the version of simulator is latest, 2020.06 for windows.

EricBoiseLGSVL, 2021-03-05:
@JM998 What do mean different?  Different coordinate system?  Different format?  We do have multiple fixes for GPS data in the next release and we hope this resolves these issues.  @hadiTab can confirm