lgsvl/simulator #1078: Adding a custom designed terrain in blender to LGSVL simulator.

siddharthjoshi01, 2020-10-20:
I have designed a custom terrain using blender 2.9, and exported the same to .fbx format in Unity. My objective is to get this terrain into the LGSVL simulator and test the terrain over a vehicle of my choice. I am working on a windows machine, and not able to find any guidelines. Need help.

EricBoiseLGSVL, 2020-10-20:
https://www.lgsvlsimulator.com/docs/add-new-map/
Let us know if you have issues ;)

siddharthjoshi01, 2020-10-20:
Yes, I have been through this video. However, it involves getting a ready-to-use downloadable asset-bundle from the asset store. In my case, I have a custom terrain designed as attached below. How to get it into LGSVL after exporting as .fbx? Will I have to create an asset-bundle myself?
[image]

EricBoiseLGSVL, 2020-10-20:
So just replace importing the asset store assets to right click in Project panel in the folder for your new map and Import New Asset...
[image]