lgsvl/simulator #1081: Save camera output issue

asifsid-32, 2020-10-21:
Hello LGSVL Team,

We are facing an issue while using save() function for saving the images from the camera sensor to our file system. Save() function here is taking more then 1 sec to write the image from the camera sensor. I want to save the images parallelly from all the four cameras such that there should not be any mismatch in the frames. 

The image frames saved from all the four cameras are not synchronised among themselves. Is there anyway possible to parallelly save the images from all the four cameras.

EricBoiseLGSVL, 2020-10-22:
@asifsid-32 You could add timestamp data into the save function and organize outside the Save instead of trying to synchronize them.  The save system is not really designed to do this at the moment.  Each sensor would need access to all others and somehow sync the save calls.  I would think just organizing the data after the fact would be easier.

asifsid-32, 2020-10-22:
> @asifsid-32 You could add timestamp data into the save function and organize outside the Save instead of trying to synchronize them. The save system is not really designed to do this at the moment. [...]

where can we find the source code for save() function ?

asifsid-32, 2020-10-22:
@EricBoiseLGSVL is it possible to reduce the delay caused in saving the images ? 
Thanks in advance.

EricBoiseLGSVL, 2020-10-22:
@asifsid-32 You can reduce the resolution of the images but other than that no.  Images are always a pain.  We are even grabbing right off the GPU to make it as fast as possible already.

asifsid-32, 2020-10-22:
> @asifsid-32 You can reduce the resolution of the images but other than that no.  Images are always a pain.  We are even grabbing right off the GPU to make it as fast as possible already.

To reduce the delay we can use opencv's imwrite() function right ? We just need frame buffer for that !!

EricBoiseLGSVL, 2020-10-22:
Yes but we can't pass the frame buffer from unity.  You could try to write a custom sensor maybe