lgsvl/simulator #1082: Unable to run carla Spawn_npc python eamples?

Amina88, 2020-10-21:
Hi I have followed the newest Carla release  following this:https://carla.readthedocs.io/en/latest/start_quickstart/#installation-summary
However when i run the spawn_npc example i have the error below:

/opt/carla-simulator/PythonAPI/examples$ python3 spawn_npc.py 
Traceback (most recent call last):
  File "spawn_npc.py", line 30, in <module>
    from numpy import random
ModuleNotFoundError: No module named 'numpy'

I have installed numpy and updated my python version yet still any ideas? 

Thank you

jollysg, 2020-10-21:
Shouldn't you be posting this issue in the Carla issues forum and not the LGSVL sim issues forum?

EricBoiseLGSVL, 2020-10-22:
Hello, welcome to LGSVL Simulator! I think you are in the wrong room :)  Our python api examples are working great on our latest release so if you don't get Carla's worked out be sure to try the latest LGSVL Simulator.

For your issue, be sure you install Python with the `-e` parameter.  If you don't, when you pull the latest Python changes, it won't update your local.  Hope this helps.

Amina88, 2020-10-22:
Hi apologies I posted in the wrong forum. However on a another note I have installed autoware and LGSVL simulator following this: https://www.lgsvlsimulator.com/docs/autoware-instructions/

But I'm trying to get  the vehicle_cmd value as when I try rostopic echo /vehicle_cmd yet values are 0 even tho the simulation is running. Any ideas ?

EricBoiseLGSVL, 2020-10-22:
@Amina88 No worries.  I see this is the issue you were having in #1045 and #1075 right?  I am going to close this issue because it is a duplicate and I will ping @hadiTab on the other issue

closed 2020-10-22 by EricBoiseLGSVL