lgsvl/simulator #1084: How to change the message type of GPS on ROS2?

francis0407, 2020-10-22:
https://www.lgsvlsimulator.com/docs/simulator-messages/#ros2-message-types mentions that the simulator supports both `nmea_msgs/Sentence` and `sensor_msgs/NavSatFix` message types for GPS.

Currently, the default message type of the GPS on ROS2 is `sensor_msgs/NavSatFix`. So, how to change this message type to `nmea_msgs/Sentence`?

dr563105, 2020-10-22:
You would need change 3 files -- Ros2Conversions, GpsData, and Ros2BridgeFactory.

[This file](https://github.com/lgsvl/simulator/blob/release-2020.06/Assets/Scripts/Bridge/Ros2/Ros2Conversions.cs#L245) has to be changed to use `nmea_msgs/Sentence`. You need to assign variables to the structures defined in [here](https://github.com/lgsvl/simulator/blob/release-2020.06/Assets/Scripts/Bridge/Data/GpsData.cs#L12). You need to change that class to reflect Sentence data type. You can either replace or add new class structure. 

Then add that class definition type [here](https://github.com/lgsvl/simulator/blob/release-2020.06/Assets/Scripts/Bridge/Ros2/Ros2BridgeFactory.cs#L40)

I think these changes are enough to help your need. There may be some other problems.

francis0407, 2020-10-23:
@dr563105  Thanks!

closed 2020-10-23 by francis0407