lgsvl/simulator #1085: Error when starting simulation: VehicleOdometryData used for publisher in ROS2 bridge

sgermanserrano, 2020-10-22:
Hi,

I'm trying to run a simulation using the lgsvl simulator and the ROS2 bridge using the `Lexus2016RXHybrid (Autoware)` Vehicle. 
I have updated the sensors on the default vehicle with the settings from [AutowareAuto lgsvl-sensors.json](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/lgsvl-sensors.json) but when trying to run the simulation I get the following error message: 
```
Invalid: Unknown data type VehicleOdometryData used for publisher in ROS2 bridge
```

Below is a screenshot of the error message.

[image]

EricBoiseLGSVL, 2020-10-22:
@hadiTab any idea?  Is VehicleOdometrySensor using old ROS2 bridge?

hadiTab, 2020-10-22:
@sgermanserrano the VehicleOdometry message isn't supported by the bridge in that release (but we will be bringing it back). If you don't care about getting vehicle odometry data you can just remove the Vehicle Odometery sensor from the vehicle's JSON configuration and things should work.

sgermanserrano, 2020-10-23:
@hadiTab thanks for the info. yeah, I just removed it for the time being.

closed 2020-10-23 by sgermanserrano

nnmm, 2021-02-02:
@hadiTab @EricBoiseLGSVL I have the same problem. When will the next release be, and is there a way to get the functionality back now (e.g. by building some development branch)?

EricBoiseLGSVL, 2021-02-02:
@nnmm You can update the bridge and add back the VehicleOdomData conversions, then make a new build.

in RosConversions.cs
```
public static Lgsvl.VehicleOdometry ConvertFrom(VehicleOdometryData data)
        {
            return new Lgsvl.VehicleOdometry() // not in 06 release
            {
                header = new Ros.Header()
                {
                    stamp = ConvertTime(data.Time),
                },
                velocity = data.Speed,
                front_wheel_angle = UnityEngine.Mathf.Deg2Rad * data.SteeringAngleFront,
                rear_wheel_angle = UnityEngine.Mathf.Deg2Rad * data.SteeringAngleBack,
            };
        }
```

VehicleOOdometryData.cs
```
namespace Simulator.Bridge.Data
{
    public class VehicleOdometryData
    {
        public double Time;
        public float Speed; // in meters per second
        public float SteeringAngleFront;
        public float SteeringAngleBack;
   }
}
```

in RosBridgeFactory.cs
```
typeof(VehicleOdometryData),
```

This is all the conversions I can find but I'll need @hadiTab to confirm.  We are working to get the release out as soon as possible.

manonechu, 2021-03-13:
Hi @EricBoiseLGSVL, I am having a similar issue, please take a look at the screenshots. I checked the RosConversions.cs, it contains the DetectedRadarObjectData. Could you please advise me? Thanks a lot!
[image]
[image]
[image]

EricBoiseLGSVL, 2021-03-14:
@manonechu what do you mean you are having similar issues?  Are the Radar msg not working with ROS2 bridge? @hadiTab is this supported?

manonechu, 2021-03-14:
@EricBoiseLGSVL, thank you for your reply. I added Radar to the JSON just like I added GPS Device and IMU on Vehicles, when I continued to the Simulations and clicked the green play button, it showed me "Invalid: Unknow data type DetectedRadarObjectData used for publisher in ROS2 bridge".

EricBoiseLGSVL, 2021-03-15:
Yes, you will need to add the bridge conversions for ROS2 since the RadarSensor does not support

manonechu, 2021-03-15:
@EricBoiseLGSVL, thank you for your reply. Would you be able to share any tutorials for adding the bridge conversions for ROS2? I was checking the RosConversions.cs, it contains the DetectedRadarObjectData already. Thanks a lot!

EricBoiseLGSVL, 2021-03-15:
Sure, we will try to get some information on how to do this.  We are trying to get the new release out, but after that we will look into more detailed docs.

manonechu, 2021-03-15:
@EricBoiseLGSVL, thank you for your rapid reply. Sounds great! I am really looking forward to it. Are there any workaround methods I can do now? Because I have to use the Radar sensor for simulation.

EricBoiseLGSVL, 2021-03-16:
sure look to the above code posted and find all references to the bridge type and add the conversions.

manonechu, 2021-03-17:
@EricBoiseLGSVL, thank you for your reply. 
I followed the above code and rebuilt it. Now it doesn't show any error when I run the simulation. But when I do "ros2 topic echo /radar", the objects array is empty. Please see the attached screenshot. I am not sure why. Do I need to turn on anything? Thanks a lot. I am looking forward to your help!
[image]

EricBoiseLGSVL, 2021-03-17:
No radar should show the vehicle in a bounding box.  This is a bug that we have fixed in the next release.  Change this:
```
var npcC = mesh.gameObject.GetComponentInParent<NPCController>();
var va = mesh.GetComponent<VehicleActions>();
```
to this
```
var npcC = mesh.GetComponentInParent<NPCController>();
var va = mesh.GetComponentInParent<VehicleActions>();
```
not sure on line numbers because I did some formatting fixes to the class

manonechu, 2021-03-18:
@EricBoiseLGSVL, thank you very much for your help!
I did the above change and rebuilt it, now I can see both vehicles in bounding boxes. However, when I did "ros2 topic echo /radar", there is nothing getting published. Before I made this change, the vehicles were not in bounding boxes, but the /radar topic got published though the objects array was empty, please see above: https://github.com/lgsvl/simulator/issues/1085#issuecomment-800736914.
Could you give me some instructions on this issue? Thank you again. I am looking forward to your reply. 
[image]

EricBoiseLGSVL, 2021-03-18:
Hmm, not sure why the data would stop sending.  Can you debug the objects array and see if the data matches the ros2 conversion methods?

manonechu, 2021-03-19:
@EricBoiseLGSVL, thank you very much for your help.
[image]
[image]
I followed your instructions by making the two changes, as shown in the above two screenshots.
The Ros2Conversions.cs seems right to me, as shown in the below screenshot. Are there any places I need to make changes? Thanks a lot!
[image]

EricBoiseLGSVL, 2021-03-19:
Can you debug the radar Detected dictionary? That way you can see if the issue is with the sensor or the bridge.

manonechu, 2021-03-19:
@EricBoiseLGSVL, thank you for your reply. What is this radar Detected dictionary for? Where can I locate this dictionary? Thank you!

EricBoiseLGSVL, 2021-03-19:
It's called detected
```
private Dictionary<Collider, DetectedRadarObject> Detected = new Dictionary<Collider, DetectedRadarObject>();
```

manonechu, 2021-03-20:
@EricBoiseLGSVL, thank you for your reply. I did a careful check but couldn't find any bugs.
When will the new version be released? Thank you!

EricBoiseLGSVL, 2021-03-22:
No problem, we are finalizing docs now, very soon ;)

MisterioRemo, 2021-03-24:
@manonechu hello!
I have the same problem with radar sensor.
In `Lgsvl.DetectedRadarObjectArray ConvertFrom(DetectedRadarObjectData data)` function in Ros2Conversions.cs I change this 
```c#
var r = new Lgsvl.DetectedRadarObjectArray()
{
   header = new Ros.Header()
  {
      stamp = Convert(data.Time),
      frame_id = data.Frame,
   },
};
```
to this
```c#
var r = new Lgsvl.DetectedRadarObjectArray()
{
   header = new Ros.Header()
  {
      stamp = Convert(data.Time),
      frame_id = data.Frame,
   },
   objects = new List<Lgsvl.DetectedRadarObject>()
};
```
And now radar topic works correctly.

EricBoiseLGSVL, 2021-03-24:
@MisterioRemo We have changed the sensor in the next release with something similar.  Thanks for the fix in the meantime, sorry for the confusion, we didn't realize this was missing.

manonechu, 2021-03-28:
@MisterioRemo, thank you! It works for me now.