lgsvl/simulator #1086: Problem with fmu vehicle

nike-96, 2020-10-22:
Hi, I followed the procedure at https://www.lgsvlsimulator.com/docs/ego-vehicle-dynamics/ to use an fmu vehicle, but I have a problem at the WebUI. After I have added the new vehicle, the simulation doesn't start sending the following error. 
[image] 
In the documentation there is nothing about the creation of a zip. I tried to put the assetbundle of the vehicle in a zip with all the files which are contained in Assets\External\Vehicles\FMU\ExampleFMU (through which there is the .dll file which is required in the error message) but another error comes up: 
[image]
I don't know what this error is about, can you help me? 
Thanks

EricBoiseLGSVL, 2020-10-22:
@nike-96 ExampleFMU.dll should be in the Resources folder.  This looks like a bug and we will look into it.  Thanks for letting us know.

nike-96, 2020-10-23:
Ok, let me know if you guys solve the issue.

nike-96, 2020-10-26:
Hi, I wanted to see the dll file so I opened it with VisualStudio and it appears like this. It seems to contain another folder (which I can't open or doing anything else); anyway it's an empty file I think and this could be the reason behind the error I get. 

[image]

nike-96, 2020-11-03:
Hi, did you manage to solve the issue?

EricBoiseLGSVL, 2020-11-04:
Sorry, still looking into it.  We have added it to the backlog to fix.  I'll update here once we have a fix

nike-96, 2021-01-14:
Any chance you solved this? Thanks

EricBoiseLGSVL, 2021-01-14:
we are still looking into it.  Sorry, next release is going to be big :)

EricBoiseLGSVL, 2021-01-14:
@nike-96 I have reviewed the code and it seems a refactor broke this.  I'll see if I can fix today or tomorrow.  I'll post code here and next release will have the fix.

nike-96, 2021-01-15:
Ok thank you!

EricBoiseLGSVL, 2021-01-15:
@nike-96 Ok all fixed.  Looks like Build.cs was missing the vehicle name folder to parse for the fmu and I also fixed a the example FMU class having odd list display issues.  Here is the patch file.  This fix will also be in the next release.  Thanks for finding this bug.

[patch.zip](https://github.com/lgsvl/simulator/files/5823270/patch.zip)

closed 2021-01-15 by EricBoiseLGSVL

nike-96, 2021-01-18:
Thank you for your support! Is that file describing what we need to change (.diff) in the old Build.cs?

nike-96, 2021-01-18:
Furthermore, do you happen to know the date of the next release?

EricBoiseLGSVL, 2021-01-19:
@nike-96 yes the diff is the changes needed.  The path is missing the name of the vehicle and the other is just fixing issues displaying the axle info in the editor.  FYI, the ExampleFMU is very rough just to show how to implement base functionality.  Release is set for Feb.