lgsvl/simulator #1088: Can not see the mesh clearly after importing fbx file.

wangludong, 2020-10-23:
Hi there
I'd like to make a annotation for my map.
But when I dragged the fbx into the scene to make a reference, I can only saw a vague contour of my mesh.

[image]

While I created a new project and dragged the same fbx, I can saw the details clearly.

[image]

So what is the problem here?

EricBoiseLGSVL, 2020-10-23:
If you are creating a mesh and importing into Simulator, materials need converted to HDRP.  When you select the mesh materials, are they pink?

wangludong, 2020-10-27:
@EricBoiseLGSVL 

Thanks for your answer.
Actually, I've already turned the lighting on before, but it had no effect.
You can see I added a Directional Light into the scene.
But I noticed that the light attributes here are quite different from those in the new project.
And when I zoomed in for a specific level, the mesh began to blur, even disappeared and only my annotation left.
[image]
[image]

EricBoiseLGSVL, 2020-10-27:
Can you open the mesh in the inspector panel to see the material and shader being used?  Is your mesh just lines?  Did you export the materials in the fbx importer and convert to HDRP materials?

wangludong, 2020-10-28:
In fact I don't have the original mesh file.
What I did is exporting the mesh from a Unreal Engine asset without choosing any option, since there are no options.
To be honest I'm not familiar with Unity, so I'm not quite sure about you mentioned here about material and shader.
From the picture below you can see the properties of the mesh in two projects.

[image]
[image]

By the way, I also take a snap of the mesh in Unreal Engine.
Hope this will be helpful.

[image]

wangludong, 2020-10-28:
Well, I find a makeshift method.
As I toggle off the scene lighting, it looks clearer.
So that's it.Thank you.

closed 2020-10-28 by EricBoiseLGSVL