lgsvl/simulator #1089: Questions regarding Apollo with lgsvl Simulator

MartinaB85, 2020-10-23:
Good morning,

I have few questions regarding the lgsvl simulator used together with Apollo. I have release-2020-06 of the simulator and both Apollo (3 and 5).

(1) I have observed that the traffic lights predictor is not working really well, as the red lights are very often not detected. This happens with both Apollo.
- Is there a way to correct this in the code? or is there a way to "cheat" meaning to skip the prediction and tell the vehicle to directly use the real value coming from the camera ? If yes, please kindly tell me which part of the code I would need to modify.

(2) Is there a way to train the Apollo vehicle in a custom scenario and then using the related AI models to drive the car?

(3) If the only way to do what I am asking in (2) is following the tutorial LaneFollowing model I already have but it does not work. When running docker-compose up drive_visual or docker- compose up collect the drive script keeps waiting for a camera image from the simulator and the sensors are correctly inserted into the vehicle. So not sure if anything else needs to be done or if there is something wrong.

Thank you in advance for your attention,

Martina

EricBoiseLGSVL, 2020-10-23:
@MartinaB85 Hello and welcome to LGSVL Simulator.
1) Yes, we are aware of this issue and still trying to debug.  We do have a work around, you can use Apollo modular testing by using the ground truth of the signals, see here #1040. You can also increase the emission values to make the lights brighter.
2) We only support AD stack EGO driving or manual driving at this time.  We hope to release automatic ego driving in a future release.
3) We are working on updating the LaneFollowing tutorial and should have this released soon

MartinaB85, 2020-10-27:
Thank you very much for your reply. I was able to fix the traffic lights perception with the signal sensor you advised me to use.

However, the car (using Apollo 5.0) is sometimes very unstable when changing lane or when driving along a straight line (see this video in attached
[Unstable_Apollo_Car.pptx](https://github.com/lgsvl/simulator/files/5445722/Unstable_Apollo_Car.pptx)
).

- Is this something related to the routing module? Are there any parameters I might need to adjust in the configuration files of the routing or any other module?

- Can Apollo be used in any costum-made scenario? Is there no need to retrain the AI models is using for classification and perception?

Thank you very much for your help,

Martina

EricBoiseLGSVL, 2020-10-27:
@MartinaB85 Glad to hear it is working.  I am not sure on the Apollo side.  You may want to post on their issues page.  @hadiTab any idea on this issue?

hadiTab, 2020-10-27:
@MartinaB85 it seems like your frame rate is pretty low. This often causes Apollo to get unstable. Using the Clock Sensor should help resolve this issue. See more information on the sensor here: https://www.lgsvlsimulator.com/docs/sensor-json-options/#clock

MartinaB85, 2020-10-28:
@hadiTab Thank you for your reply. 

(1) I added the clock sensor to the Apollo 5.0 vehicle but as soon as I modify the file cyber.pb.conf adding clock_mode: MODE_MOCK below run_mode: MODE_REALITY, **Dreamview fails to start**. I also tried rebuilding Apollo after modifying the cyber.pb.conf file but Dreamview still fails to load. Any hint on this?

(2) I might try running Apollo and the simulator on two different machine later on but for now **I would need to use my machine**, which is quite powerful. It's i7 Intel Core with 8 processors and 32 GB of RAM. I was going through this issue https://github.com/lgsvl/simulator/issues/769 because the same happens to me. When running Lgsvl simulator on my machine which has Ubuntu 18.04 all 8 processors are almost at 100%. I tried lowering the rendering quality in Unity but the result is the same. I have checked few things (if the machine uses GPU for rendering, the frame rate etc) and everything seems to be fine to me. Is it a problem related to the linux version? Did you find a solution to this problem?

Thank you very much for helping me,

Martina

EricBoiseLGSVL, 2020-10-28:
@MartinaB85 What GPU are you using?

MartinaB85, 2020-10-28:
@EricBoiseLGSVL  I'm using the GeForce GTX 1070 with the Nvidia driver version 450.80.02

EricBoiseLGSVL, 2020-10-29:
@MartinaB85 2) Yes, it seems a few Linux machine users have this issue.  We have not been able to reproduce yet.  We plan on updating our Unity version soon and hopefully it solves this issue.