lgsvl/simulator #1092: How should I register topics with the simulator side through ros2 lgsvl bridge?

CHENxh2017, 2020-10-27:
On the ROS2 side, I write a ROS2 node that subscribes to topics or publishes topics. How should I get the simulator to interact with those topics you need to register them with the simulator side?

hadiTab, 2020-10-27:
What are the message types for these topics? Are they custom message types? If they are standard ROS 2 message types it is likely that the simualtor already supports them.

It would be much easier to walk you through this if you could provide more details on what you are trying to do. If your messages have custom message types you would need to either directly add them to the source code in the simualtor, or you could build a plugin that adds functionality for that message type. However, the simulator won't really do anything with that data. You will need to create a "sensor" in the simulator that interacts with the message in some way.

Thinsheep, 2021-05-06:
Hi @CHENxh2017  
Are you familiar with this topic now? Have you found any examples to publish/subscribe topics to/from the bridge?