lgsvl/simulator #1093: is there a semi-truck available

manonechu, 2020-10-27:
Hi, thank you for the great simulator.

I would like to know that is there a semi-truck available? Because I'd like to simulate autonomous driving for a semi-truck. If it is not available, is there a way to create or import it? Please kindly advise me.

Thank you so much for your help!

EricBoiseLGSVL, 2020-10-27:
@manonechu  Thanks!  We are working on supporting semi-trucks but it is not ready yet.  You can create a NPC and then attach Unity FixedJoint components to the trailer.  You will also need to create a new LaneFollowBehaviour so you can adjust turn radius and behavior for this new NPC.  We hope to release examples soon.

manonechu, 2020-10-27:
Thank you for your response.
I really look forward to seeing the support for semi-trucks in the further
release and the examples you mentioned in the email. I will keep following.

EricBoiseLGSVL, 2020-10-27:
@manonechu No problem.  Be sure to reach out with any issues if you attempt to add a semi-truck on your own.

manonechu, 2020-10-29:
Thank you, Eric. Would it be easier to add a conventional truck than a
semi-truck? I attached a screenshot of the conventional truck below.
Thanks a lot.
[image: image.png]

EricBoiseLGSVL, 2020-10-29:
I don't see the image but yes a conventional truck would be much easier.  You can create a new npc and add it to the simulator internals.  We will also support external NPCs very soon.  The issue with a rig is that the fixed joint can be difficult to work with in Unity,

manonechu, 2020-10-30:
@EricBoiseLGSVL , thank you again for your help, Eric. I attached the figure again. I meant the conventional truck marked by the red rectangle. Because I took this figure from your [tutorial](https://www.lgsvlsimulator.com/docs/npc-map-navigation/), does it mean that you already have the support for it and I can add it from somewhere? Or I need to follow the tutorial [here](https://www.lgsvlsimulator.com/docs/add-new-ego-vehicle/) to create it by myself? 
Thanks a lot.
[image]

EricBoiseLGSVL, 2020-10-30:
Cool, no this tutorial is for EGO vehicles.  You will need to add an NPC.  You will need to look at the other npcs and copy their setup and add the the internal list of npcs.  There is a issue about it here #755

manonechu, 2020-11-01:
@EricBoiseLGSVL, I really appreciate your generous help. 
I am thinking to design an autonomous driving system for BoxTrucks, i.e., I want to control the steering, throttle, and brake of BoxTrucks, via C++ and ROS. Would you please advise me on how to simulate the autonomous system in the LG simulator? Some steps would be great!

The BoxTruck is one of NPC vehicles. I would like to know if I can also control the steering, throttle, and brake of NPC vehicles, just like controlling Ego vehicles.

EricBoiseLGSVL, 2020-11-02:
@manonechu Yes you can use the box truck as an EGO vehicle.  I would look at the tutorial on how to add a new ego vehicle, use the NPC box truck, modify the mesh and make a new EGO.  Then you can control with ROS.  I'm not sure why you need to use C++, do you need a new vehicle control sensor?  You can make it in C# as a sensor plugin.

manonechu, 2020-11-03:
@EricBoiseLGSVL 
Thank you very much, Eric. That would be very helpful for me.
I can also go with Python plus ROS. But C++ seems to be a better language for development as Autoware and others are using it.
Please kindly let me know if you have an update of the tutorial on how to add a box truck as a new ego vehicle. I will pay attention to your online document, too.
Thank you again.

EricBoiseLGSVL, 2020-11-03:
@manonechu no sorry, this is the docs that will be updated on the next release.  This is the new tutorial [here](https://youtu.be/R_Z65kimCII)

manonechu, 2020-11-04:
@EricBoiseLGSVL , thank you very much, Eric. Hope you have your favorite president tonight.

manonechu, 2020-11-10:
@EricBoiseLGSVL, Hi Eric, when I set up the simulator and want to control the vehicle, why the type of topic "/vehicle_cmd" is "autoware_msgs/VehicleCmd"? How could I use the lgsvl/msgs to control the vehicle? I attached two screenshots of the simulator below.
Thank you so much for your help.
[image]
[image]

EricBoiseLGSVL, 2020-11-10:
@hadiTab do you know why the topics are different?

hadiTab, 2020-11-10:
@manonechu this looks like a bug. I'm going to have to review it in more detail later, but it seems like right now whenever the bridge type is set to ROS it will expect the `autoware_msgs` control command.

manonechu, 2020-11-10:
Thank you, Hadi, will it work if I have a ROS node in which publishes
commands via the autoware_msgs?

Or do I need to use ROS2 instead to use the lgsvl/msgs?

Thanks a lot for your help.

manonechu, 2020-11-12:
@hadiTab Thank you, Hadi, will it work if I have a ROS node in which publishes
commands via the autoware_msgs?

Or do I need to use ROS2 instead to use the lgsvl/msgs?

Thanks a lot for your help.

hadiTab, 2020-11-13:
@manonechu both of those options would work.

manonechu, 2020-11-13:
@hadiTab thank you for your reply. In ROS, I publish commands via the autoware_msgs, but it seems that the vehicle only responses to the twist_cmd. Does it mean that I cannot directly control the steering angle?
When I select ROS2 in the vehicle setting in WebUI, as shown in the following picture, I can see the lgsvl/msgs and I can rostopic pub /vehicle ..., but the vehicle does not response.
[image]
I install ROS2 but I don't see the /vehicle topic when I ros2 topic list.
Could you please advise me?  Thanks a lot.

hadiTab, 2020-11-13:
@manonechu you are right. You need to use the /twist_cmd to control the vehicle using `autoware_msgs`. I believe the steering angle is mapped to the z-axis of the angular velocity part of the `twist_cmd` message. However, there is a bug in the Autoware control that makes it a bit hard to control the velocity precisely and because of that I would recommend using ROS2.
In ROS2 you can just use the [lgsvl_msgs](https://github.com/lgsvl/lgsvl_msgs) to control the car. It should just subscribe to the `VehicleControlData` message.

manonechu, 2020-11-13:
@hadiTab thank you very much for your reply.
When using ROS, I ran the [rosbridge](http://wiki.ros.org/rosbridge_suite) and published commands via autoware_msgs (twist_cmd).
My question is, when using ROS2, what rosbridge should I install/use? Right now, I have rosbridge from ROS1 running and use ROS2 to publish lgsvl_msgs, but the vehicle doesn't respond. And I type "ros2 topic list", it doesn't show any topics from the simulator.
Thank you for your help again.

manonechu, 2020-11-14:
@hadiTab thank you very much for your help.
I follow this [tutorial](https://www.lgsvlsimulator.com/docs/build-instructions/) to build LG simulator and this [tutorial](https://www.lgsvlsimulator.com/docs/ros2-bridge/) to set up the ROS 2 bridge.
I have an error to connect to the simulator. Could you please advise me on how to solve this issue? Thank you very much.
This is the vehicle:
[image]
This is the simulation:
[image]
This is the error I have:
[image]

hadiTab, 2020-11-14:
@manonechu It looks like you are using an older version of the simulator that does not support the ros2-lgsvl-bridge. Did you build your own binary? If so, what branch did you use? The ros2-lgsvl-bridge only works with the 2020.06 release. You can download a prebuild binary for it from here: https://github.com/lgsvl/simulator/releases/tag/2020.06
If you need to build your own binary, make sure to checkout the `release-2020.06` branch before building.

manonechu, 2020-11-14:
@hadiTab Thank you very much for your help. I rebuilt the release-2020.06 and now the ros2-lgsvl-bridge can talk to the simulator.
I tried to publish the topic /vehicle_cmd to control the steering angle, throttle, and brake. However, the vehicle only responded to the /acceleration_pct and /acceleration_pct, but **not** to target_wheel_angle and target_wheel_angular_rate. Please see the below figure. 
Could you please advise on this issue? Thanks a lot.
[image]

hadiTab, 2020-11-14:
Have you tried bigger numbers? I think the angle is in degrees (but not sure).

manonechu, 2020-11-14:
@hadiTab Thank you very much for your quick reply.
I increased the /target_wheel_angle to bigger values and the vehicle turned the front wheels! Thank you.
I also noticed that the vehicle will only turn the front wheels when the /target_wheel_angle is large than 0.68. And the front wheels will turn to **the same** position as long as the /target_wheel_angle is large than 0.68.
Please let me know if I am mistaken. 
Here is the [lgsvl_msgs](https://github.com/lgsvl/lgsvl_msgs/blob/master/msg/VehicleControlData.msg).

hadiTab, 2020-11-15:
@manonechu This is a bug that will be fixed in the next release. Sorry I forgot about it.
Look at [this comment](https://github.com/lgsvl/simulator/issues/1065#issuecomment-710689578) for a fix.

manonechu, 2020-11-17:
@hadiTab Thank you very much! I fixed the bug and the steering control works now.

closed 2021-07-16 by EricBoiseLGSVL

Bmoradi93, 2021-11-02:
@EricBoiseLGSVL 
Is the tractor-trailer system integrated into the vehicle library?
Please give us an update on this.
My MSc Thesis is designing a local planner for tractor-trailer systems in backward maneuvering.
I would like to use lgsvl in my research for the simulation part.
Thank you!

EricBoiseLGSVL, 2021-11-03:
@Bmoradi93 This feature is not yet finished but you can create a trailer system using the new articulated body components from unity.

Bmoradi93, 2021-12-03:
@EricBoiseLGSVL 
The problem is I do not know if it will be compatible with ROS2 (Getting topics and etc in ROS2 like the Lexus vehicle)
I'm done with my theoretical and python implementation. 
We would like to migrate from Gazebo to unity and looking for someone to help us in this direction.
His/her name will be included in our papers.
If you know someone, please introduce if possible.

EricBoiseLGSVL, 2021-12-03:
You could start an issue and I can mark it help wanted.  Describe what help you need and maybe others can contact or post