lgsvl/simulator #1095: Is it possible to reset the environment using gym lgsvl env based on out of road boundary condition ?

Revbr, 2020-10-27:


hadiTab, 2020-10-27:
@Revbr you would need to manually right the code to check this. i.e. check vehicle position from GPS against road boundary in HD map and reset if you are out of bounds. It would require a bit of work to search the HD map for the right road segment.

Revbr, 2020-10-28:
Thanks for the reply. How to get the HD map details in gym lgsvl env or what is the python API to be called for getting HD map details?

hadiTab, 2020-11-03:
Unfortunately Python API doesn't provide a way to access HD map details. You would need to download the HD map and write Python Code for reading it (or find open-source tools for the specific type of HD map you are using).