lgsvl/simulator #1096: Traffic light not recognized in San Francisco map

tgif73, 2020-10-27:
Hello,
I'm using 2020.06 simulator with Apollo 5.0. I'm seeing traffic light is not recognized correctly (always shows Green) in San Francisco map. Please see screenshot - Traffic light module shows Red light but it says Green.

[image]

I tried the Modular Testing as well, added 3D Groud Truth Sensor and Signal Sensor to the vehicle, it has other issues. The vehicle runs really slow and stops at traffic light whether it's Green or Red. And the vehicle won't move again.

However, when I use other map, like Shalun, Apollo detects the traffic light perfectly fine (and stops at red light and runs at green light). Also, the Modular Testing works fine as well in other map.
All maps are downloaded from lgsvl content link (https://content.lgsvlsimulator.com/maps/)

Is there anything I can to fix these issues?

Thank you for the help in advance.

EricBoiseLGSVL, 2020-10-27:
@luqiang21 Did we find the fix for this?

EricBoiseLGSVL, 2020-10-27:
@tgif73 Please see #1089

tgif73, 2020-10-29:
Hi @EricBoiseLGSVL  so the traffic light recognition works fine with Sensor Signal enabled in EGO. However, the EGO stops at traffic light (whether it's green or red). Is there a workaround to fix this issue? Also, this issue only happens in San Francisco map. Other map, GoMentum, works find in Modular testing mode (everything works as expected)
Will this be fixed in next release? When is the next release?

Thank you~!

EricBoiseLGSVL, 2020-10-30:
@tgif73 I don't think we have the signal ids setup in SF for 06 release.  It should be updated in our next release.  This will allow signal sensor to work

lemketron, 2020-10-31:
> I tried the Modular Testing as well, added 3D Groud Truth Sensor and Signal Sensor to the vehicle, it has other issues. The vehicle runs really slow and stops at traffic light whether it's Green or Red. And the vehicle won't move again.

I believe the problem here may be that you are running Apollo perception and traffic light modules, though you said you are using Modular Testing, which is where the simulator provides 3D ground truth and traffic signal information which is intended to be used in place of Apollo's Perception and Traffic Light modules.

Can you try this again but do not enable Perception or Traffic Light in the Apollo module controller?

tgif73, 2020-11-01:
Hi @lemketron yes, I did disable perception and traffic light modules from Apollo. With 3D Ground Truth and Signal sensors added to vehicle parameters in LGSVL, I do see correct traffic light states (RED/YELLOW/GREEN) however, the EGO stops at traffic light intersection and won't move again.

lemketron, 2020-11-01:
> Hi @lemketron yes, I did disable perception and traffic light modules from Apollo. With 3D Ground Truth and Signal sensors added to vehicle parameters in LGSVL, I do see correct traffic light state changes (RED/YELLOW/GREEN) however, the EGO stops at traffic light intersection and won't move again.

Where do you see the traffic light state changes?  Can you locate the actual signal in cyber-monitor? The traffic signal (camera) perception window (upper left corner) should not appear if those modules are disabled. I see that there is also "GREEN" showing in dreamview which is good, and there is a fence with an hourglass showing in front of the car (and there is no path showing in front of the car either). 

When this happens, it can help to disable the Planning module, wait a bit (usually the switch will flip back on and then eventually settle in the off position) and then re-enable planning. At this point if you have already set a route in the Route Editing tab and clicked "send route" then the path should appear in front of the car and since Control is enabled, the car should begin to drive.

tgif73, 2020-11-01:
I see the traffic light state changes in Dreamview's upper right side (next to Auto). It changes from RED GREEN YELLOW correctly (perception window doesn't appear since I have Traffic Light module being disabled).
I tried disable/enable Planning but the PATH doesn't show. If I start the EGO from LGSVL with keyboard arrow key, EGO starts to move and the PATH is shown again.

Also how do you monitor signal value from cyber monitor? All I see is list of signals and its frame ratio...

[image]

lemketron, 2020-11-01:
In cyber monitor, use the arrow keys to move the selection bar down to /apollo/perception/traffic_light and hit return. You should then see a list of traffic_light objects (usually one or zero) and can expand that to see the color. 

[image]

The red line indicates the desired route, but I'm not sure why you still have the signal stop fence when the light is green.

It looks like you're using the LGSVL fork of Apollo 5.0 so please do a "git pull" to ensure that you have the latest Apollo HD map files which need to have signal IDs that match the signal IDs in the latest Simulator environments, or Apollo may not recognize the signal when it changes.  You might also want to try removing the simulator folder (`~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/`) (to force a fresh download of the latest simulator maps).

Have you tried this on the Borregas or CubeTown maps to see if the signals are recognized there?

I apologize for the confusion here; we had to update signal IDs (to match) everywhere for this feature and it can be a little tricky to troubleshoot when it doesn't work.