lgsvl/simulator #1097: Finding absolute coordinate in the map

tgif73, 2020-10-27:
Hello,

What's the proper way of finding absolute coordinate in the map (for lgsvl.Vector function)? San Francisco map is open source so I can open the map in Unity's editor mode and place an object onto where I want in the map to get absolute coordinate. However, other maps (with no open source) like GoMentum, I am not sure how to find the absolute coordinate in the map. I've drive around the EGO vehicle with GPS on, but I'm not sure how to convert those GPS coordinate into absolute map coordinate.

Thank you for the help.

EricBoiseLGSVL, 2020-10-27:
@tgif73 Do you mean Unity world position?  If so, you can use the api to place a ego, pedestrian, controllable at 0,0,0 and this will be the origin point.

tgif73, 2020-10-27:
Thanks for the reply. I mean to create waypoints with absolute position (not relative to EGO position), how can I find the x/y/z coordinates? For example, if I want to spawn a NPC at some corner, how do I find that coordinate?

Thank you

EricBoiseLGSVL, 2020-10-27:
@tgif73 Yep, that is Unity world position. If you look at AddAgent.cs in the source, setting the position is in world coordinates.  It's easy to tell from any code that is object.transform.position.  This is for EGO, NPC, Pedestrian and Controllables.  To find the position in the maps that are not open source, just make a new sensor (plugin or in Editor) and output the position as you drive around.  Then record the position to spawn.  We are working on a visual scenario editor that will release soon that will open non open source maps and let users position agents in the map, set waypoints and triggers.

tgif73, 2020-10-28:
Thank you @EricBoiseLGSVL . So, there is no prebuilt sensor that will give the world coordinate. I'll follow the instruction to create a custom sensor and try it out. Thanks.

martins-mozeiko, 2020-10-28:
There are two alternatives:

1) you can load map with PythonAPI and then use [sim.map_from_gps](https://www.lgsvlsimulator.com/docs/python-api/#helper-functions) function to convert lat/lon to map coordinates. To get lat/lon position for maps like GoMentum, you can use Google maps or similar online service.

2) you can run simulator from inside of Unity Editor. Once the map loads, you can pause the simulation with Unity pause button, and then place new object in scene, position wherever you want it and write down coordinates from Unity inspector panel.

tgif73, 2020-10-28:
Thank you @martins-mozeiko . The second option works great for my need - I didn't know we can load binary map in Editor mode.

closed 2020-10-28 by tgif73

duongtran11, 2021-05-12:
> Thank you @EricBoiseLGSVL . So, there is no prebuilt sensor that will give the world coordinate. I'll follow the instruction to create a custom sensor and try it out. Thanks.

In the second option: why dont we just create an object in the editor and drag it to where we want to know the coordinates, the coordinates in x, y, z in inspector panel. no need to run then pause the simulator

marcusvinicius178, 2021-09-27:
@martins-mozeiko @EricBoiseLGSVL I need to standardize the goal spot to send to the EGO vehicle reaches in Autoware and in Apollo.Auto. My map is GoMentum.

As I am comparing different trajectory planners from the 2 ADS (Apollo/Autoware) I need to send the vehicle in the same goal spot ALWAYS. How can I do that, considering that Dreamview on Apollo and Rviz on Autoware.AI enable me send the vehicle just using a manual icon. 

And also considering that the coordinates on these frameworks (Dreamview;Autoware) are **not the same as lgsvl coordinates (LAT/LONG)**

On Dreamview Apollo I have:

[image]

If you see the coordinates on bottom right (image) are totally different from where the car is in LGSVL (take a look in terminal the LAT=37 LONG=-121 and on Dreamview-Apollo LAT=592761 and LONG=4134480)

On Autoware.AI I have the option to send the goal spot through command line, but just send the vehicle to go ahead x meters, and reache a determined orientation....
But is not possible to send to a specific LAT/LONG. 

It would be nice to know how is possible to standardize these coordinates, or to understand how to fix some goal spot to enable the car-robot always goes to these **SAME** coordinates for every path planning test.

Some idea, suggestion?

EricBoiseLGSVL, 2021-09-28:
@marcusvinicius178 Did you find a solution?