lgsvl/simulator #1098: not running with ubuntu

kotakky, 2020-10-28:
I installed unity hub and unity() in order to launch default simulation.
However, Many Error occurred as shown below image .As a result,I cannot render the map in  unity with linux  
Please help me What I should do.

[image]
[image]

EricBoiseLGSVL, 2020-10-29:
Did you enable git lfs before you cloned?

kotakky, 2020-11-04:
Yes,git lfs is installed.
However, The errors are not resolved.   
[image]

kotakky, 2020-11-04:
In addition, when I enter playmode, the below message is received .
"All compiler errors have to be fixed before you can enter playmode!"

martins-mozeiko, 2020-11-04:
Did you install & initialize git-lfs before cloning repository? Because git lfs does its work during fetch/checkout process, not aftwerwards.

Try running `git lfs ls-files`. All files should have asterix (`*`) next to name. If you see dash (`-`) that means you did not have git lfs properly configured when cloning repository. Run `git lfs fetch && git lfs checkout` to get LFS files in proper state.

https://www.lgsvlsimulator.com/docs/faq/#why-are-assets-scenes-missing-empty-after-cloning-from-git

kotakky, 2020-11-04:
When I run git lfs ls-files, nothing is shown (Not in a git repository) .
[image]

EricBoiseLGSVL, 2020-11-04:
open terminal in the root location simulator in cloned and then run the commands @martins-mozeiko has posted please.  You can also tell if files are not pulled from git lfs if you navigate to Assets/UI/Images and see that the files are not images but text files in the Unity editor.  Do this for External cloned repos as well (vehicle/maps/etc)

kotakky, 2020-11-05:
Thank you for your support.
git lfs  can work properly.However, I get the below message.I want to know how to address that.
[image]

EricBoiseLGSVL, 2020-11-05:
You need to use Vulkan, OpenGL is not supported for Linux

kotakky, 2020-11-05:
As you said, Simulator say Vulkan on the top of the Unity Editor.
But, the Shalun's scene does not colorize.
[image]
[image]

martins-mozeiko, 2020-11-05:
By default scenes do not have any light in them. Light is added only when scene is loaded in simulation during normal run - this is done so light system is common across all scenes.

If you want to see scene in the editor, disable lighting - click lightbulb icon next to `2D` text in the toolbar.

kotakky, 2020-11-05:
I clicked the bulb icon and so, the scene is like below.
But I got the message "NO camera rendering".
Is this message normal?I think I sholud resolve the problem.
[image]

EricBoiseLGSVL, 2020-11-05:
Yes, this is fine.  The camera is added at runtime but you are trying to load the map without loading simulator objects.  To test in editor you need to build the bundle, run Loader.scene, create a new simulation and map in WebUI (point map to new created bundle in repo/AssetBundles or where ever you placed it). Then select run the new simulation with the new map.  This will let you run in editor.  We do have Developer Settings in Simulator menu in between Window and Help at the top.  This will let you run the simulator from the new map scene you are working on but it limited in scope.  Next release we will have more options with this feature.

closed 2020-11-18 by kotakky