lgsvl/simulator #1099: Issue with Unity 2019.3.15f1 on Ubuntu 20.04

nike-96, 2020-10-28:
Hi, I am having textures problems with Unity. The version I am using is 2019.3.15f1 and I am running on Ubuntu 20.04. The problem is in the visualization of the vehicle, which appears like this: 

[image]

Could it be a problem of compatibility between Unity's version and Ubuntu's one? Or a problem of the git clone of the car (Jaguar2015XE) taken from github? I am totally lost. 
Thanks

hadiTab, 2020-10-28:
I think this is just lighting. Try toggling the lighting button at the top of the scene view (it's the button with a lightbulb on it).

EricBoiseLGSVL, 2020-10-29:
[image]

closed 2020-11-10 by EricBoiseLGSVL