lgsvl/simulator #1100: Integrate the Radar Model available in Matlab with LGSVL simulator ?

Jiaqing-Cao, 2020-10-28:
Hi, I am researching some radar simulator works, Matlab has some sensor detections which could run 3D on Unreal engine.

I woul like to know, is it possible to integrate the Radar Model available in Matlab with LGSVL simulator ?

If possible, could you please share some ideas ? Thank you!

EricBoiseLGSVL, 2020-10-29:
@Jiaqing-Cao Yes this is possible.  You can create a sensor plugin that communicates with matlab

Jiaqing-Cao, 2020-10-29:
@EricBoiseLGSVL 
Oh, that's great.

Could you please share some more info about how to create a sensor plugin that communicates with matlab?
I looked the tutorials here[https://www.lgsvlsimulator.com/docs/sensor-plugins/](url).
But it seems that is one tutoiral for adding Unity created sensor plugins to LGSVL.

EricBoiseLGSVL, 2020-10-29:
@mark-gerow-lge When you have time can you explain the basics for creating sensor plugins, thanks.

mark-gerow-lge, 2020-10-29:
@Jiaqing-Cao if you follow the tutorial for a sensor plugin at 

https://www.lgsvlsimulator.com/docs/sensor-plugins/

you can create custom C# scripts for unity that allows you to collect data from the ego vehicle / simulation, then send it to Matlab via a socket connection. Then you can build the plugin, and use it in your simulations and attach to your ego vehicle. Make sure to inherit from SensorBase with your actual sensor (helper scripts do not have that restriction) and make sure its on the root object of your prefab.

Jiaqing-Cao, 2020-10-29:
@mark-gerow-lge ,
Thanks a lot for the explanation! 
I will try that.

Jiaqing-Cao, 2020-11-06:
Hi, @EricBoiseLGSVL @mark-gerow-lge .
I have been researching Matlab with LGSVL for some time.
I look into [mathworks](https://www.mathworks.com/help/driving/ref/simulation3dsceneconfiguration.html).
It seems for now, Simulation 3D Scene Configuration only support Unreal Engine.

Could you please share some ideas about that ?

EricBoiseLGSVL, 2020-11-06:
You will need to look at the code that this package provides and create similar functionality for Unity.  It really shouldn't be that much different from Unreal.  We are working on something similar already,  I found this [link](https://www.mathworks.com/matlabcentral/answers/196774-connection-between-matlab-and-unity3d) that might help.

Jiaqing-Cao, 2020-11-06:
Okay, thank you for that!