lgsvl/simulator #1101: About Python API bug

CurlyWei, 2020-10-29:
Hi LGSVL team,
I use the version of 2020.06 and find two bug.

1.windshield_wipers
2.turn_signal_left

I type the code ,but it didn`t work.
> ego = sim.add_agent("Jaguar2015XE_Autoware", lgsvl.AgentType.EGO, state)
> c = lgsvl.VehicleControl()
> c.windshield_wipers = 3
> c.turn_signal_left = True   # bool
> ego.apply_control(c, True)

How do i need to modify?

Thanks

EricBoiseLGSVL, 2020-10-29:
@CurlyWei We do not support wipers on vehicles since we migrated to HDRP.  You would need to add this to the API commands in simulator, add vehicle animations and code to toggle them and add the Python API code.  Turn signal should work fine.  We will check if this is a bug.

EricBoiseLGSVL, 2020-10-30:
@shalinmehtalgsvl Do you know if this (turn signal not wipers) is an issue in latest?

shalinmehtalgsvl, 2020-10-31:
Hi @CurlyWei ,

This is an issue in the latest build as well. Thank you for reporting the issue, we're looking into it.

The keyboard controls should still work `,` turns on the left blinker, `.` turns on the right blinker, and `m` turns on both (hazards).