lgsvl/simulator #1103: Simulation with Apollo, Vehicle don't move?

duongnxptit, 2020-10-29:
I have release-2020-06 of the simulator and try with Apollo follow guide :
- [ Running with latest Apollo ](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/) 
- [ youtube tutorial ](https://www.youtube.com/watch?v=Ucr0aM334_k&t=1610s&ab_channel=LGSVLSimulator)
All are good until select routing in Apollo then submit its. Vehicle don't run autonomous, there is only one red line that gives directions to the selected point.

luqiang21, 2020-10-29:
@duongnxptit There is only one red line means `routing` works but `planning` is not working. You need to make sure `perception` and `prediction` are working correctly. Also, can you let me know what is your GPU memory? Normally limited memory will cause Apollo perception fails.

duongnxptit, 2020-10-30:
@luqiang21, thanks for you reply! I am using RTX 2060 6Gb. I have capture my CPU and GPU when app running. I think this problem with CPU because it load too high

lemketron, 2020-11-01:
You can lighten the load on the GPU memory by leaving the Perception and Traffic Light modules off in DreamView and using our Modular Testing feature instead where 3D ground truth sent from the simulator eliminates the need to use the Perception and Traffic Light modules in Apollo.

Refer to the [Modular Testing Documentation](https://www.lgsvlsimulator.com/docs/modular-testing/) and [Modular Testing Video](https://youtu.be/781zkzN2xMg) for more information.

duongnxptit, 2020-11-01:
> You can lighten the load on the GPU memory by leaving the Perception and Traffic Light modules off in DreamView and using our Modular Testing feature instead where 3D ground truth sent from the [...]

Thanks for your advice, i have tried Modular Testing, it worked very well. So my problem is GPU memory is not enough for run Perception and Traffic Light. Is there a way to run Perception and Traffic Light modules in Apollo, without Modular Testing? feature?

luqiang21, 2020-11-01:
@duongnxptit One possible way is to try to run apollo in one computer and run simulator in another. Simulator can be run using window or linux.

duongnxptit, 2020-11-02:
Thanks for your help, @luqiang21 . I have run simulator in another windows computer. Everything is working fine. eg routing, perception, traffic light

closed 2020-11-02 by duongnxptit