lgsvl/simulator #1105: Timestamp on image

asifsid-32, 2020-10-29:
Hello lgsvl team,

I am trying to synchronise the camera output image generated from save() function. How can we print the timestamp on camera output image, which will help us to synchronise the images from multiple cameras for the same instances.

EricBoiseLGSVL, 2020-11-02:
@asifsid-32 @hadiTab What do you think? Custom msg?

hadiTab, 2020-11-02:
@asifsid-32 this isn't exactly a timestamp, but you could use the PythonAPI call `sim.current_time` to get the current simulation time in seconds. You can call this right before saving the image and it should give you a timestamp you can use (technically there will be one frame delay between the timestamp and the recorded image though).