lgsvl/simulator #1109: Unity Editor enters an unresponsive black screen with the simulator

nniranjhana, 2020-11-04:
I followed the instructions on https://www.lgsvlsimulator.com/docs/build-instructions/ and when I click on the project simulator, it loaded up a black screen on the Unity Editor and I'm unable to do step 12, as this window (attached image below) won't close (but it minimizes) and it is unresponsive if I click on any options (for Simulator > Build WebUI..).

I tried creating a new project (not adding the cloned simulator) and it worked fine. I rebooted and I still had the same problem. How do I fix this issue?

[image]

EricBoiseLGSVL, 2020-11-04:
First off download and install Unity 2019.3.15f1 for the 06 release [here](https://unity3d.com/get-unity/download/archive).  Be sure to include Windows modules.
Close unity and delete the Library folder in the root of the cloned repo.  Open Unity and let it create the folder.  Close Unity after it is done and open it again.
If you still have issues, please find the Editor player log [here](https://docs.unity3d.com/Manual/LogFiles.html) and post.

nniranjhana, 2020-11-05:
I followed your instructions, unfortunately the problem still persists. PFA the Editor log.

[Editor.log](https://github.com/lgsvl/simulator/files/5496093/Editor.log)

EricBoiseLGSVL, 2020-11-05:
This is your issue
```
Vulkan API version 1.1.0 (1.2.131 supported by driver)
Vulkan vendor=[Intel] id=[8086]
Vulkan renderer=[Intel(R) UHD Graphics 630 (CML GT2)] id=[9bc5]
Vulkan device type 1
Vulkan driverversion=[0x05000008] uint=[83886088]
Vulkan PSO: cache data not found [path=/tmp/LG Silicon Valley Lab/LGSVL Simulator/vulkan_pso_cache.bin]
Initialize mono
```

You need to be sure Simulator is using the GPU.  Please see the linked issues here #1057

nniranjhana, 2020-11-05:
Thanks for pointing me to the root of the issue! I checked #1057, #1076, #879, #753 which were all related. But I could not find the exact solution there (I had already added the PPA, my NVIDIA drivers installed, `libvulkan1` and `vulkan-tools` installed (but vkcube didn't render) and I did not want to run LGSVL in a docker container to solve this).

So on Ubuntu, I had to open NVIDIA X Server Settings, go to PRIME Profiles tab, and choose NVIDIA On-Demand (it was previously in Performance Mode) and then reboot. I am now able to work with the simulator on a responsive Unity Editor.

For anyone else who might need it in the future:

https://askubuntu.com/questions/1201072/how-nvidia-on-demand-option-works-in-nvidia-x-server-settings

[image]

EricBoiseLGSVL, 2020-11-06:
Awesome and thanks for posting your solution for others.  You rock

closed 2020-12-02 by EricBoiseLGSVL