lgsvl/simulator #1111: Using LGSVL with Apollo over ssh

LukeBauer, 2020-11-05:
I've been trying to run Apollo and LGSVL with a connection over ssh since I don't have an in network computer that can handle both of them. Currently the connection can be made, but commands are extremely stuttered if they are even registered at all. Is it even possible to run this connection over ssh or is the bandwidth too small?

EricBoiseLGSVL, 2020-11-05:
Yes, I feel this is not enough bandwidth.  For Apollo we recommend a gigabit switch, lower than that, Apollo does not work with simulator.

martins-mozeiko, 2020-11-06:
It will work fine over ssh connection as long as you can push 10-12 MB/s over it real-time (that's approximate bandwidth used by default set of sensors Apollo). Latency also might be an issue, don't know exact numbers, but I would guess anything more than 30-50 msec will be unusable.

lemketron, 2020-11-09:
If you are able to run Apollo 6.x (master), then you can consider using "[Simulator time](https://www.lgsvlsimulator.com/docs/sensor-json-options/#clock)" (and `MODE_MOCK`) to combat potential latency issues.