lgsvl/simulator #1112: how to deal with mismatch between map annotation and actual road?

Addi2020, 2020-11-06:
as you can see in the picture there is a a little slop in actual road (shown in point cloud data).
All the traffic lanes (map annotation), point cloud render and 3D plan are align (having same height, y-axis ), but the map annotation can been seen clear in some part (left side of picture) but can not be seen on the right side. Which means the actual road having slop.

Now the question is After map annotation When i upload the hd map to apollo dreamview, There is a mismatch between hd map and actual road. 
[image]
[image]

luqiang21, 2020-11-06:
I think you need to have correct mesh based on the point cloud with actual height instead of a plane. The annotations won't have correct height if your mesh doesn't have. Once you got the correct mesh, you can use following button to snap annotations to the road.
[image]

Addi2020, 2020-11-07:
@luqiang21 , Sorry i didn't understand about this mesh? (I think you need to have correct mesh based on the point cloud with actual height instead of a plane.) could you please explain how to do it?

luqiang21, 2020-11-07:
@Addi2020 I mean you should have mesh collider to be correct as in real world, not 
>  3D plan are align (having same height, y-axis )

Otherwise, you can't have correct height for the map annotations.

Addi2020, 2020-11-08:
@luqiang21 got it. Thanks!

closed 2020-12-16 by Addi2020