lgsvl/simulator #1113: Collecting data of vehicle

nike-96, 2020-11-06:
Hi, I'd like to collect data of the vehicle after/in real time with the simulation. I'm referring to vehicle position, velocity, orientation, acceleration and maybe something more specific reguarding vehicle dynamics like tire wear or slip angle. Is it possible? How can I do? 
If I need to create a sensor, can you tell me the procedure? Do I need to use Apollo/Autoware or not? 
Thank you

EricBoiseLGSVL, 2020-11-10:
@nike-96 Please read the documentation [here](https://www.lgsvlsimulator.com/docs/).
Look at the sensors info on what sensors collect what data for position, vel, orientation, acceleration etc.
To collect data on features we don't have, you will need to create sensor plugins in the sim source or as a plugin.  Since Unity has a limited wheel collider features, you may need to extend these.  There is also a great Unity asset called Vehicle Dynamics Pro that might help show you what is needed but we do not support at this time.
Sensor plugins [here](https://www.lgsvlsimulator.com/docs/sensor-plugins/)
I recommend using an AD stack to control the vehicle but it is not required.

nike-96, 2020-11-12:
Ok thaks. 
I dowloaded Vehicle Physics Pro and tried to create an assetbundle of the SportCoupe. 
At first it required a VehicleInfo component and I simply copied and pasted the one of the Jaguar2015XE; after that, I managed to create the assetbundle. 
The problem is in the simulation though. In the WebUI, it sends a message of the following type: 

[image]

What could be the problem? 

I have another question. I need to switch from Windows to Linux in order to use Apollo. I could use only Linux since Unity and lgsvl are both supported, but I am not an expert and I still prefer windows: for this reason I'd like to work as much as possible on windows and when Apollo is required simply switch to Linux. I was wondering if the assetbundles which have been generated on windows can be seen and used in the WebUI opened on Linux. Is it so?

EricBoiseLGSVL, 2020-11-12:
@nike-96 You will need to make a new simulator exe with the new VPP classes included.  You also may need to edit the build process to include the VVP attached classes.  I would need to see the error from building the bundle and or the error in WebUI.

Yes, we required Windows and Linux build modules to make an asset bundle.  They are cross platform.  (be sure you have both installed to make the previous VPP bundle)

nike-96, 2020-11-13:
Hi, I have tried to make a new simulator.exe but the error in the webui is still the same one as before (the one referring to an "object reference not set to .." in the first post). 
If I remove the vehicle info component that I added (taking the one of the Jaguar as previously said) the error in the console of Unity Editor is the following: 

//
Exception: Build failed: Vehicle info on Assets\External\Vehicles\vpp_car\vpp_car.prefab not found. Please add a VehicleInfo component and rebuild.
Simulator.Editor.Build+BundleData.PreparePrefabManifest (Simulator.Editor.Build+BundleData+Entry prefabEntry) (at Assets/Scripts/Editor/Build.cs:197)
Simulator.Editor.Build+BundleData.RunBuild (System.String outputFolder) (at Assets/Scripts/Editor/Build.cs:296)
Simulator.Editor.Build.OnGUI () (at Assets/Scripts/Editor/Build.cs:613)
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <437ba245d8404784b9fbab9b439ac908>:0)
Rethrow as TargetInvocationException: Exception has been thrown by the target of an invocation.
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) (at <437ba245d8404784b9fbab9b439ac908>:0)
UnityEditor.HostView.Invoke (System.String methodName, System.Object obj) (at <a20a50bbfaff4193bf6ba6b262f9a71f>:0)
UnityEditor.HostView.Invoke (System.String methodName) (at <a20a50bbfaff4193bf6ba6b262f9a71f>:0)
UnityEditor.HostView.InvokeOnGUI (UnityEngine.Rect onGUIPosition, UnityEngine.Rect viewRect) (at <a20a50bbfaff4193bf6ba6b262f9a71f>:0)
UnityEditor.DockArea.DrawView (UnityEngine.Rect viewRect, UnityEngine.Rect dockAreaRect) (at <a20a50bbfaff4193bf6ba6b262f9a71f>:0)
UnityEditor.DockArea.OldOnGUI () (at <a20a50bbfaff4193bf6ba6b262f9a71f>:0)
UnityEngine.UIElements.IMGUIContainer.DoOnGUI (UnityEngine.Event evt, UnityEngine.Matrix4x4 parentTransform, UnityEngine.Rect clippingRect, System.Boolean isComputingLayout, UnityEngine.Rect layoutSize, System.Action onGUIHandler, System.Boolean canAffectFocus) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, UnityEngine.Matrix4x4 worldTransform, UnityEngine.Rect clippingRect, System.Action onGUIHandler, System.Boolean canAffectFocus) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, System.Action onGUIHandler, System.Boolean canAffectFocus) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, System.Boolean canAffectFocus) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.IMGUIContainer.SendEventToIMGUI (UnityEngine.UIElements.EventBase evt, System.Boolean canAffectFocus) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.IMGUIContainer.HandleEvent (UnityEngine.UIElements.EventBase evt) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.CallbackEventHandler.HandleEventAtTargetPhase (UnityEngine.UIElements.EventBase evt) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.MouseCaptureDispatchingStrategy.DispatchEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcher.ApplyDispatchingStrategies (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, System.Boolean imguiEventIsInitiallyUsed) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcher.ProcessEventQueue () (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcher.OpenGate () (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcherGate.Dispose () (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.EventDispatcher.Dispatch (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, UnityEngine.UIElements.DispatchMode dispatchMode) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.BaseVisualElementPanel.SendEvent (UnityEngine.UIElements.EventBase e, UnityEngine.UIElements.DispatchMode dispatchMode) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.UIElements.UIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at <78b213560771414e9fa6f1b95f5ad8bb>:0)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at <819de1aa368e45faa4f78e26c97c62b0>:0)
//

What do you mean by "edit the build process to include the VVP attached classes"? 

Furthermore, I have just noticed that the car doesn't appear so well in Unity Editor: with the lighting on, it shows like this: 

[image]

Thaks

EricBoiseLGSVL, 2020-11-13:
- You need the VehicleInfo class on the root prefab or you need to change simulator code to exclude it.
- You need to edit the build process if it is throwing errors attaching the VPP components. (Post these please)
- If you are getting vehicle errors when building the simulator, only have the simulator checked in build process.  Build one asset at a time and make sure sim build is not in dev mode build
- The reason your vehicle is pink is that you did not convert the materials to HDRP.  VPP is not compatable with HDRP so there are steps needed to update.  This might mean code needs updated also.  Do you have console errors with VPP in simulator editor before even running the editor?

nike-96, 2020-11-16:
- Yes, I need to add VehicleInfo component. I think adding the one which is present in the Jaguar2015XE prefab is correct, isn't it?Anyway, if I do so the building process seems to be fine (at the end of the building the console prints "done" and, despite of a bunch of warnings, no errors appear. As I try to start the simulation on the WebUI though, the same message appears ("Object reference not set to an intance of an object"). 

- The VPP components are:

[image]

- The problem of the pink vehicle has been solved, by converting the materials to HDRP as you told me!

nike-96, 2020-11-16:
I'm sorry, I forgot to mention that the picture shows the components of the VPP vehicle at the download of the asset (so without the VehicleInfo component that I added later on).

lemketron, 2020-11-16:
@nike-96 You might also refer to https://github.com/lgsvl/simulator/issues/515 in case it helps.

EricBoiseLGSVL, 2020-11-16:
Please open the asset bundle and screen shot the contents

nike-96, 2020-11-17:
I have added the Assetbundle browser tool. The problem is that if I go into "Insector" panel (on Window > Assetbundle Browser) I can't inspect the assetbundle of the vpp vehicle that I created; the strangest thing is that this holds even for all the other assetbundles that I created following the lgsvl's documentation! It just says "Invalid Assetbundle". What could it be? 

[image]

Please recall that the other assetbundles which have been created from the original lgsvl prefab work well on the simulator and doesn't create any problem even if, as I just said, they are "invalid" in the inspector panel in the assetbundle browser. 

[image]

nike-96, 2020-11-17:
I have just found out that using Python API to run the simulator I am able to load the VPP car, but I receive the same error I get in the WebUI: 
[image]

Despite of the error, the car appears in the scene and I am able to drive it, but I can't exploit  API functionalities like spawning in certain points, callbacks and so on..

EricBoiseLGSVL, 2020-11-18:
No just open asset bundle with a zip application.  Look with explorer etc.  Don't use the Unity AssetBundle package

nike-96, 2020-11-18:
I tried to open it with Visual Studio but the result is this: 
[image]
Is it encrypted?

EricBoiseLGSVL, 2020-11-18:
no you just need a zip program.  7-zip or 8-zip or any free zip software

nike-96, 2020-11-19:
Ok, here you have both the Jaguar and VPP car assetbundles: 
[image]
[image]

nike-96, 2020-11-19:
By the way I am not able to open any C# scripts of VPP vehicle. Furthermore they have a different icon with respect to scripts of lgsvl vehicles. I post them below: 
- VPP scripts 
[image]

-lgsvl scripts 
[image]

I want to open them in order to see if I can edit them properly, but I can't do so: as I double click on the scripts in the inspector panel VisualStudio doesn't start. Do you know what could be the issue? 
The reason behind this is in https://github.com/lgsvl/simulator/issues/515 where the user create new scripts for the vehicle to be able to inherit from IVehicleDynamics. Is it still so? Does any vehicle in the simulator need to inherit from IVehicleDynamics.cs? If so, I need to edit the scripts just like in that post (maybe) to solve my problem.

EricBoiseLGSVL, 2020-11-20:
Sweet OK yes these look correct.  I think we will need to test locally to see what the issue is.  Yes if you make a new Dynamics class it needs to inherit IVehicleDynamics. Just a new Dynamics class, not anything else.  If you replace VehicleSMI.cs

EricBoiseLGSVL, 2020-12-02:
@nike-96 did you get a vehicle with VPP classes working?

nike-96, 2020-12-03:
No. The thing is that all dll files of VPP would have to go through a process of reverse engineering which would require a big amount of time and furthermore without the certainty of being succesful..

EricBoiseLGSVL, 2020-12-03:
@nike-96 Oh it's a .dll file?  Yes, you will not be able to edit those.  Does the .dll have any exposed parameters for throttle and steering?  You could make a dumb IVehicleDynamics that connects to the .dll and applies the forces.  That way you have the required dynamics but it only communicates with VPP.

EdyJ, 2020-12-07:
The Community Edition of VPP is provided as DLL. It includes standard Unity components and API, but only the source code is not exposed. There's no need for reverse-engineering, as everything is documented, for example:

https://vehiclephysics.com/components/vehicle-controller/
Scroll down to the Scripting Reference section for more details on the exposed API.

All input parameters (throttle, brake, steering etc) may be sent to the vehicle easily using the exposed Data Bus:

https://vehiclephysics.com/advanced/databus-reference/

The data bus is also an excellent source for querying the actual state of the vehicle (engine rpm, gear, speed, etc).

I'm also working on an advanced telemetry system on-par with high end car analysis tools. However, this one may take some tome to land in the Community Edition:

https://twitter.com/VehiclePhysics/status/1335953312944951296

EricBoiseLGSVL, 2020-12-07:
@EdyJ Thanks for your help.

BBlumi97, 2021-04-22:
@nike-96 Did you got a car with VPP Dynamics running ? I am also interested in doing it.

nike-96, 2021-04-22:
No I have not tried anything new since I was not interested in VPP anymore, sorry.

BBlumi97, 2021-04-22:
okay thanks anyway!

closed 2021-07-16 by EricBoiseLGSVL