lgsvl/simulator #1115: Failure in running route editing

Yimags, 2020-11-07:
Hello guys,

I am using the lastest apollo stack together with lgsvl in modular test. After building it from scratch, I test it for several time and it works well at beginning. However, from time to time the ego vehicle is initilized with the stop sign and even a wall with it, which leads to failure in using the route editing function. Any suggestion to solve it? Thanks in advance!
[image]

ntutangyun, 2020-11-07:
take a look at the planning module output in the `cyber_monitor`. see if there are any errors or warnings

GabrielZZZ, 2020-11-08:
I have also encountered this issue. The cybver_montor shows that the navigation module is down. My temporary solution is that you click _Reset _All button in the screenshot you showed and then re-enabled those modules. Hopefully, the car will start to move.

I suppose the reason is that Apollo believes it is driving a real car, and such a real car will not "suddenly moves back to the original start".

Yimags, 2020-11-08:
> cyber_monitor

Here is the screen shot of monitor, do you mind telling me how I could tell if it works correctlly and how to fix it? Thx!
[image]

Yimags, 2020-11-08:
> take a look at the planning module output in the `cyber_monitor`. see if there are any errors or warnings

Here is the screen shot of monitor, do you mind telling me how I could tell if it works correctlly and how to fix it? Thx!
[image]

Yimags, 2020-11-08:
> I have also encountered this issue. The cyber_monitor shows that the navigation module is down. My temporary solution is that you click _Reset All button_ in the screenshot you showed and then re-enabled those modules. Hopefully, the car will start to move.
> 
> I suppose the reason is that Apollo believes it is driving a real car, and such a real car will not "suddenly moves back to the original start".

Yeah, I have tried this method as well. It doesn't work from time to time. And the situation occurs as well when I try to interupt during the automated navigation.

EricBoiseLGSVL, 2020-11-10:
We are looking into a solution for this but yes @GabrielZZZ is right, Apollo is not expecting the vehicle to warp :)

Yimags, 2020-11-10:
> We are looking into a solution for this but yes @GabrielZZZ is right, Apollo is not expecting the vehicle to warp :)
Thx! now I understand resetting the ego vehicle could make trouble. But this situation (stop sign) also occurs when I tried to deviate from the allocated route during automatic navigation. Does it result from the same problem that not yet solved?

EricBoiseLGSVL, 2020-11-10:
This could be the same issue but Apollo should account for deviation.  @hadiTab @luqiang21 Any idea?

luqiang21, 2020-11-10:
@Yimags I didn't see such cases when I interrupt the vehicle during automatic navigation. In your screenshot, the red line (route) is behind the ego vehicle, do you know why? The route should appear in front of the ego. And check more in cyber_monitor and logs when planning doesn't work.

Yimags, 2020-11-10:
> @Yimags I didn't see such cases when I interrupt the vehicle during automatic navigation. In your screenshot, the red line (route) is behind the ego vehicle, do you know why? The route should appear in front of the ego. And check more in cyber_monitor and logs when planning doesn't work.

@luqiang21 Sorry for the misleading screenshot above. Here are some error cases I encountered with similar problem.
1.Crossing the terminal of last navigation and failing to start next navigation
[image]
2.Deviating from the allocated route and failing to assign next target
[image]

luqiang21, 2020-11-11:
@Yimags Sorry I also don't have much idea on these situations. For 1, I guess the ego might have gone a bit further than the starting location of the next navigation. Thus planning is failing to generate a trajectory. Maybe you can try to set the routing again? For 2, I also had similar experience that the planning won't work if the ego is far from lane center. That seems something to tune in apollo planning module. I don't have better solution except manually moving the ego closer to the lane.

lemketron, 2020-11-11:
You could try posting this issue over at the Apollo Issues page:
https://github.com/apolloauto/apollo/issues

closed 2020-12-02 by EricBoiseLGSVL