lgsvl/simulator #1116: Unknown Sensor Type Comfort

SuriyaNitt, 2020-11-07:
Hi,

I am trying to add the example Comfort Sensor Plugin.

While I am able to build the custom sensor plugin successfully, launching the simulation with the custom sensor throws error saying that the sensor is of unknown type.

I checked to make sure that the prefab and the 2 other c# scripts are present as part of the Assets/.../External/ComfortSensor folder.

The build AssetBundle is < 20 KB. 

Any help is greatly appreciated.

EricBoiseLGSVL, 2020-11-10:
@mark-gerow-lge the comfort sensor issues are resolved in the next release correct?  Is there anything you can suggest to resolve the issue locally?
@SuriyaNitt Please post the error (player.log) to help us see the exact error

mark-gerow-lge, 2020-11-16:
Hey @SuriyaNitt thanks for your paitence, can you send me your built Comfort Sensor Plugin so I can take a look at what went wrong? When I build one for myself it ends up ~18kb so the size sounds roughly correc.t Also just to confirm, there were no warnings or errors in your console log when building?

mjshiggins, 2020-11-20:
Just wanted to say that I'm seeing the same thing on my end. Excerpt from Player.log:

> Exception: Unknown data type ComfortData used for publisher in ROS2 bridge
>   at Simulator.Bridge.BridgeInstance.AddPublisher[DataType] (System.String topic) [0x00066] in <2b0a300bc6084ea2ab7b92800f231ba7>:0
>   at Simulator.Sensors.ComfortSensor.OnBridgeSetup (Simulator.Bridge.BridgeInstance bridge) [0x00007] in <ec1b003e0fcd479ba8b85053bb3c5712>:0
>   at SensorsController.InstantiateSensors (System.String sensors) [0x001d5] in <e59dc819ed674a28a0165157489d7750>:0
>   at SensorsController.SetupSensors (System.String sensors) [0x0001b] in <e59dc819ed674a28a0165157489d7750>:0
>   at AgentManager.SpawnAgent (Simulator.AgentConfig config) [0x00131] in <e59dc819ed674a28a0165157489d7750>:0
>   at Simulator.Api.Commands.AddAgent.Execute (SimpleJSON.JSONNode args) [0x00281] in <e59dc819ed674a28a0165157489d7750>:0
>   at Simulator.Api.ApiManager.Update () [0x0010a] in <e59dc819ed674a28a0165157489d7750>:0
> UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
> UnityEngine.DebugLogHandler:LogException(Exception, Object)
> UnityEngine.Logger:LogException(Exception, Object)
> UnityEngine.Debug:LogException(Exception)
> Simulator.Api.ApiManager:Update()