lgsvl/simulator #1117: Problems with perception and traffic light modules

GabrielZZZ, 2020-11-08:
Hi guys,

I have a problem about running the LGSVL simulator with the **latest** Apollo program. The fps of LGSVL simulator and Dreamview is larger if I using **Modular testing** that bypasses the Apollo's perception modules. However, I just want to **test the performance of the real perception module, but it turns on and off all the time.** I have two Nvidia RTX 2080ti in my computer so I think it may not be a GPU's problem. All the versions of the components have met the requirements in the documentation. The system version is Ubuntu 20.04 LTS.

Does anyone have ideas about how to improve the performance of the simulator and Dreamview, particularly make the perception and traffic light modules behave as expected?

Thanks a lot!:wink:

ntutangyun, 2020-11-09:
could you try to launch the perception modules in the terminal with `cyber_launch` and see what errors are showing when they shuts down.

GabrielZZZ, 2020-11-11:
Hi @ntutangyun, can you provide the exact command? I cannot find the information on the website about the path behind `cyber_launch start`...

Thanks!

lemketron, 2020-11-11:
@GabrielZZZ This might help:
http://www.fzb.me/apollo/howto/how_to_run_perception_module_on_your_local_computer.html

GabrielZZZ, 2020-11-11:
Thanks @lemketron . Here is the screenshot for me to run the command `cyber_launch start /apollo/modules/perception/production/launch/perception_all.launch`:
[image]