lgsvl/simulator #1118: Controlling Speed of Ego Vehicle

asifsid-32, 2020-11-09:
Hello LGSVL Team,

I want to control speed of ego vehicle such that when i am using the keyboard as input it shouldn't cross a specified maximum speed. Even though i am pressing the forward key it should follow the specified speed. Is there anyway to do it without using pythonAPI ?

ntutangyun, 2020-11-09:
you could add a custom check on the ego vehicle's speed when it's reading the forward key. if the speed exceeds the threshold, maybe you could simply ignore the forward key.

asifsid-32, 2020-11-09:
> you could add a custom check on the ego vehicle's speed when it's reading the forward key. if the speed exceeds the threshold, maybe you could simply ignore the forward key. 

@ntutangyun in which file we are supposed to make changes ??

ntutangyun, 2020-11-09:
hi @asifsid-32 maybe this is what you want:

https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Dynamics/Examples/VehicleSMI.cs#L498
```c#
        if (VehicleController != null)
        {
            SteerInput = VehicleController.SteerInput;
            AccellInput = VehicleController.AccelInput;
        }
        
        if (RB.velocity.magnitude > 5f) 
        {
            AccelInput = 0f;
        }

```

asifsid-32, 2020-11-09:
> hi @asifsid-32 maybe this is what you want:
>
>
>
> https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Dynamics/Examples/VehicleSMI.cs#L498
>
> ```c#
>
>         if [...]

Thanks !!

hadiTab, 2020-11-09:
@asifsid-32 if you want to always drive at a constant speed you can add the [cruise control sensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#cruise-control).

closed 2020-12-02 by EricBoiseLGSVL