lgsvl/simulator #1119: Increasing maximum FieldOfView of Lidar sensor

BaltashovIlia, 2020-11-09:
Hello LGSVL Team, 

As I see in documentation, maximum FieldOfView is currently 45 degrees but what about other lidars with higher vertical field of view, like Ouster OS-0, RoboSense RS-Bperl and Hesai PandarQt?

I am trying to set FieldOfView greater when maximum allowed and getting incorrect point cloud. Here are some videos of the problem: https://drive.google.com/drive/folders/1AHafykfL0x1Jc_0QnXH_v-JaTW-hWIuu?usp=sharing 

To reproduce the incorrect point cloud copy this sensor configuration yaml and start San Francisco world: 
```yaml 
[
     {
        "type": "Lidar",
        "name": "Velodyne VLP-32C",
        "params": {
          "LaserCount": 32,
      "FieldOfView": 90.0,
      "CenterAngle": 40,
          "MinDistance": 0.5,
          "MaxDistance": 10,
          "RotationFrequency": 10,
          "MeasurementsPerRotation": 1800,
          "Compensated": true,
          "PointColor": "#ff000000",
          "Topic": "/point_cloud",
          "Frame": "velodyne"
        },
        "transform": {
          "x": 0,
          "y": 2.312,
          "z": -0.3679201,
          "pitch": 90,
          "yaw": 90,
          "roll": 0
        }
    },
{
    "type": "Vehicle Control",
    "name": "AD Car Control",
    "params": {
      "Topic": "/vehicle_cmd"
    }
},
{
    "type": "Keyboard Control",
    "name": "Keyboard Car Control"
}
]
```

The same problem can be reproduced with CenterAngle greater than allowed:  
```yaml 
[
     {
        "type": "Lidar",
        "name": "Velodyne VLP-32C",
        "params": {
          "LaserCount": 32,
      "FieldOfView": 40.0,
      "CenterAngle": 65,
          "MinDistance": 0.5,
          "MaxDistance": 10,
          "RotationFrequency": 10,
          "MeasurementsPerRotation": 1800,
          "Compensated": true,
          "PointColor": "#ff000000",
          "Topic": "/point_cloud",
          "Frame": "velodyne"
        },
        "transform": {
          "x": 0,
          "y": 2.312,
          "z": -0.3679201,
          "pitch": 90,
          "yaw": 90,
          "roll": 0
        }
    },
{
    "type": "Vehicle Control",
    "name": "AD Car Control",
    "params": {
      "Topic": "/vehicle_cmd"
    }
},
{
    "type": "Keyboard Control",
    "name": "Keyboard Car Control"
}
]
```

And even setting FieldOfView to 44 and CenterAngle to 44, I got an incorrect point cloud: https://drive.google.com/file/d/1o9uESsuHYwdpvT50u7KkASC9WxZWRuQp/view?usp=sharing 

Additionally, what about lidars with limited horizontal angle, like IBEO lidars?

EricBoiseLGSVL, 2020-11-10:
We don't recommend increasing the fov greater than 45.  This issue explains why #559

BaltashovIlia, 2020-11-10:
Hi, thanks for the answer. 

But, do you have any plans to change how the lidar camera is rendered? I think it should be done because sensors with fov greater than 45 are available now. As I wrote before, these are Ouster OS-0, RoboSense RS-Bperl and Hesai PandarQt. And even Velodyne has plans for the lidar with grater fov – VelaDome. 

I cannot use two simulator sensors to represent one lidar, because I want to implement RoboSense RS-Bperl in simulation and it has a vertical fov from 0 to +90 degrees.

EricBoiseLGSVL, 2020-11-10:
@BaltashovIlia No problem.  We do have plans to improve and simulate real world sensors but we have a big release with major changes to simulator releasing soon and have not been able to work on this yet.  This is definitely on our roadmap.

closed 2020-11-15 by BaltashovIlia