lgsvl/simulator #1120: Ground Truth Visualizer is not working

ideasplus, 2020-11-10:
Hi, I want to visualize bounding boxes of  the obstacles as detected by the AD stack which running perception module. According to the official documentation, I add the **3D Ground Truth Visualizer sensor** and **2D Ground Truth Visualizer sensor** to the vehicle sensor configuration. However, I can't see the detected obstacles on the simulator as shown in the figure.

[image]

EricBoiseLGSVL, 2020-11-10:
Enable the 3D Ground Truth and 2D ground truth in the visualize panel

lemketron, 2020-11-10:
It looks like you enabled the (2D and) "[3D Ground Truth Visualizer](https://www.lgsvlsimulator.com/docs/sensor-json-options/#3d-ground-truth-visualizer)" sensors which are intended to visualize bounding boxes as supplied by an external AD stack. 

You need to instead add and enable the "[3D Ground Truth](https://www.lgsvlsimulator.com/docs/sensor-json-options/#3d-ground-truth)" sensor and enable visualization for that sensor. Sorry for the confusion, but please refer to the linked docs (above) for more details on each.

ideasplus, 2020-11-11:
@EricBoiseLGSVL @lemketron
Thank you for your replay.  But the `3D Ground Truth` and `2D Ground Truth` sensors return 3D and 2D ground truth data without running perception module. What I want to do is **visualize the output of the perception module** instead of ground truth data.  

Here is the screenshot without running any AD stack, I only run the simulator  and enable the  3D Ground Truth and 2D Ground Truth sensors.
[image]

lemketron, 2020-11-11:
> What I want to do is **visualize the output of the perception module** instead of ground truth data.

Sorry, I missed that. In that case, I believe you need to publish to the specified topic name (the example in the documentation is `/simulator/ground_truth/3d_visualize`), or change the topic name to match the topic from your AD stack.

ideasplus, 2020-11-11:
@lemketron 
>  or change the topic name to match the topic from your AD stack.

Hi, I'm using Apollo, I want to know which apollo's topic should be matched. I change `/simulator/ground_truth/3d_visualize` to `/apollo/perception/obstacles`, but it does't work. Correct me if I'm wrong. Looking forward to your reply.

EricBoiseLGSVL, 2020-12-02:
@lemketron is this the topic you use?

lemketron, 2020-12-02:
This is the correct topic, but unfortunately that sensor was created for autoware and has not yet been updated to support apollo perception output. In addition, it should be renamed as it is intended to be a "perception visualizer" not a "ground truth visualizer" and now that we have a "ground truth sensor" it is even more confusing.

I have filed a bug internally to fix the name and also add support for visualizing apollo perception output, but cannot promise when this will be addressed. Sorry for the confusion!

ideasplus, 2020-12-02:
@EricBoiseLGSVL @lemketron  Thanks for your clarification. Looking forward to the new feature.

lemketron, 2021-07-06:
Please refer to the Apollo Perception Visualizer 3D sensor here:
https://www.svlsimulator.com/docs/simulation-content/sensors-list/#apollo-perception-visualizer-3d

closed 2021-07-16 by EricBoiseLGSVL