lgsvl/simulator #1122: Problems with Apollo 6.0 on LGSVL

1Jyfc, 2020-11-10:
Hi team!
I'm working on Apollo 6.0 recently, and I tried to simulate Apollo on LGSVL. The hardware and software configurations are follow:

- CPU: Xeon(R) W-2123 CPU @ 3.60GHz
- Graphics card: Nvidia TITAN Xp
- Ubuntu version: 18.04
- Nvidia version: 450.80.02
- Cuda version: 11.0
- Apollo version: v6.0, commit 7ed7f7d0cb9232a5ce2b219e77c3c81b72d68e20

Here are the problems I have met:

1. The one hour long video on Youtube about how to simulate Apollo on LGSVL is great. I launched APollo and LGSVL following the video and documents on official website, however there're some problems:
- I can't see all obstacle-perception results in Dreamview. Some of them are visible, while some are not. Besides, traffic light switch in Dreamview can't be opened. I tried to launch it in Dreamview and by the following code:
`cyber_launch start /apollo/modules/perception/production/launch/perception_all.launch`
but both failed.
- For prediction, the prediction trajectories of those visible obstacles don't show. Here is the output of `cyber_monitor`:
[image]
It seems that obstacle-prediction is not launched.
2. I can run simulation with problem 1. Obviously the car can't respond to surroundings. However, even though there's no car around it, when going straight, the car will wag left and right. It seems that the car wants to return to the trajectory and turns left, but it will run too left then turn right to return again. Finally it wags more and more violent.

In summary, I'm hoping for your help. Could you give me the hardware and software configurations in the demo video (since it succeeded)? Or is there any demo on the latest Apollo 6.0? Thanks a lot. Sorry to bother.

EricBoiseLGSVL, 2020-11-10:
@hadiTab @luqiang21 Any ideas on this?

luqiang21, 2020-11-10:
@1Jyfc To me it looks like not enought GPU memory. Do you have many sensors in your ego vehicle? You can monitor gpu usage when you run both the simulator and apollo to confirm.

lemketron, 2020-11-10:
> 1. However, even though there's no car around it, when going straight, the car will wag left and right. It seems that the car wants to return to the trajectory and turns left, but it will run too left then turn right to return again. Finally it wags more and more violent.

This is usually a sign of low frame rate due to the CPU being overloaded. One way to solve this is to run the Apollo stack on a separate machine from the simulator, and another way to solve it is to switch Apollo 6.0 to use `MODE_MOCK` using the simulated [clock sensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#clock). This will help apollo process the data as it arrives (according to the simulator time) instead of assuming the data was missed, and should solve the weaving issue.

1Jyfc, 2020-11-16:
@luqiang21 @lemketron thanks for your reply. Sorry that I reply so late because of some other things. These days I tried to run Apollo and LGSVL on two machines, one windows 10 and another ubuntu 20.04, and here is my problem:

On Apollo 5.0, which is a version that LGSVL previded on  [lgsvl/apollo-5.0](https://github.com/lgsvl/apollo-5.0), the result is satisfying. The predicted trajectories of obstacle vehicles are shown on Dreamview, and the weaving issue was solved too. (Though it seems that the traffic light module still has something wrong, with the car ignoring the red light).

However, on Apollo 6.0 (commit `7ed7f7d0cb9232a5ce2b219e77c3c81b72d68e20`), there's something wrong on Dreamview. With the same configuraion as 5.0, Dreamview still cannot show the trajectories. And the code:
`cyber_launch start /apollo/modules/perception/production/launch/perception_all.launch`
is also useless. `nvidia-smi`shows that:
[image]
It seems OK.

LGSVL and Apollo-6.0 are connected normally. The two machines are in the same public network, and they can ping each other. Is there any suggestion? Hope for reply, thanks.

Here are some pictures about my configuration and problems:
LGSVL configuration:
[image]

A scene of LGSVL and Dreamview. There is no obstacle on Dreamview while all the modules are running.
[image]
[image]

1Jyfc, 2020-11-17:
Hello! I'd like to add some new experiment results.

I tried to use 3D ground-truth and signal sensor as [LGSVL Simulator Demo: Apollo modular testing](https://www.youtube.com/watch?v=781zkzN2xMg&feature=youtu.be) shows. It runs almost OK:
[image]

Something wrong is that when given a destination, the car will move for a moment and then stop. As the video shows:
[image]

There will be a red wall when red light shows. And in my case, the red wall will appear for a while, and then disappears. The car will move while it appears and stop when it disappears and return to the state that picture 1 shows.

Here is my vehicle configuration:
`[
  {
    "type": "3D Ground Truth",
    "name": "3D Ground Truth",
    "params": {
      "Frequency": 10,
      "Topic": "/apollo/perception/obstacles"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
},
{
    "type": "Signal",
    "name": "Signal Sensor",
    "params": {
      "Frequency": 10,
      "MaxDistance": 100,
      "Topic": "/apollo/perception/traffic_light"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
},
  {
    "type": "CAN-Bus",
    "name": "CAN Bus",
    "params": {
      "Frequency": 10,
      "Topic": "/apollo/canbus/chassis"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS Device",
    "name": "GPS",
    "params": {
      "Frequency": 12.5,
      "Topic": "/apollo/sensor/gnss/best_pose",
      "Frame": "gps"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS Odometry",
    "name": "GPS Odometry",
    "params": {
      "Frequency": 12.5,
      "Topic": "/apollo/sensor/gnss/odometry",
      "Frame": "gps"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "GPS-INS Status",
    "name": "GPS INS Status",
    "params": {
      "Frequency": 12.5,
      "Topic": "/apollo/sensor/gnss/ins_stat",
      "Frame": "gps"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "IMU",
    "name": "IMU",
    "params": {
      "Topic": "/apollo/sensor/gnss/imu",
      "Frame": "imu",
      "CorrectedTopic": "/apollo/sensor/gnss/corrected_imu",
      "CorrectedFrame": "imu"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": -1.348649,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Radar",
    "name": "Radar",
    "params": {
      "Frequency": 13.4,
      "Topic": "/apollo/sensor/conti_radar"
    },
    "transform": {
      "x": 0,
      "y": 0.689,
      "z": 2.272,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Keyboard Control",
    "name": "Keyboard Car Control"
  },
  {
    "type": "Wheel Control",
    "name": "Wheel Car Control"
  },
  {
    "type": "Vehicle Control",
    "name": "Apollo Car Control",
    "params": {
      "Topic": "/apollo/control"
    }
  }
]`

I noticed that the json in video is a little different from [LGSVL Apollo 5.0 vehicle](https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/) in control part. I'm not sure if it matters, since it's also wrong when I change it as the video shows.

I think this experiment may show that almost everything except perception module of Apollo is all right, and the CPU and GPU status of the two machines are normal. In Apollo repository of Github there're also some issues related to this problem (like [this issue](https://github.com/ApolloAuto/apollo/issues/12998). So could you provide me the version of Apollo in the [guide video](https://www.youtube.com/watch?v=Ucr0aM334_k) or the [module test video](https://www.youtube.com/watch?v=781zkzN2xMg&feature=youtu.be)? And I'd like to know if you succeed on the latest Apollo. Thank you a lot!

ntutangyun, 2020-11-22:
Maybe you could check the planning output with `cyber_monitor` and see why the ego is not moving.

vatanaksoytezer, 2020-12-17:
@1Jyfc  I think the car stops because of the traffic light being red, which is indicated in top left of the dreamview's red barrier. One thing you can try is turning the light green via PythonAPI or from the parameters in LGSVL (should be left of plug sign)